betaflight/betaflight-configurator #3207: 10.9.0-RC3 : "No dataflash chip found" using cloud build option "FLASH"

MaryJane136, 2023-01-04:
### Describe the bug

Using cloud build option "FLASH" no dataflash available for blackbox logging.
When using "Classic Mode" blackbox logging is possible.

Status "Cloud Build":

MCU F745 Clock=216MHz, Vref=3.30V, Core temp=38degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 3719, max available: 32768
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1 locked dma
GYRO=MPU6000, ACC=MPU6000, MAG=HMC5883
OSD: MSP (53 x 20)
BUILD KEY: 4.4.0-RC3/IFLIGHT_F745_AIO/dae8eb48740108d1714b56214a306beb
System Uptime: 10 seconds, Current Time: 2023-01-04T10:13:45.966+00:00
CPU:27%, cycle time: 115, GYRO rate: 8695, RX rate: 15, System rate: 9
Voltage: 5 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
Arming disable flags: RXLOSS CLI MSP RPMFILTER

Status "Classic Mode":

MCU F745 Clock=216MHz, Vref=3.30V, Core temp=38degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 4218, max available: 32768
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1 locked dma
GYRO=MPU6000, ACC=MPU6000, MAG=HMC5883
OSD: MSP (53 x 20)
BUILD KEY: 4.4.0-RC3/IFLIGHT_F745_AIO/2b7332b3d02bbd9735314f2e36311537
System Uptime: 15 seconds, Current Time: 2023-01-04T10:06:23.016+00:00
CPU:25%, cycle time: 124, GYRO rate: 8064, RX rate: 15, System rate: 9
Voltage: 5 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: Not configured
FLASH: JEDEC ID=0x00efaa21 128M
Arming disable flags: RXLOSS CLI MSP RPMFILTER

### To Reproduce

Use cloud build options and select option "FLASH".

### Expected behavior

Onboard dataflash chip should be available.

### Configurator version

10.9.0-RC3 (norevision)

### Flight controller configuration

```txt
diff all

# version
# Betaflight / STM32F745 (S745) 4.4.0 Jan  3 2023 / 23:37:31 (2c34b3949) MSP API: 1.45

# config: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name IFLIGHT_F745_AIO
manufacturer_id IFRC
mcu_id 004a00274b53501220393848
signature 

# name: Bumblebee HD V3

# resources
resource MOTOR 5 NONE
resource MOTOR 6 NONE

# feature
feature -ESC_SENSOR
feature GPS

# serial
serial 0 1 115200 57600 0 115200
serial 5 0 115200 57600 0 115200

# beacon
beacon RX_LOST
beacon RX_SET

# led
led 0 0,0::C:4
led 1 1,0::C:4
led 2 2,0::C:4
led 3 3,0::C:4
led 4 4,0::C:4
led 5 5,0::C:4
led 6 6,0::C:4
led 7 7,0::C:4
led 8 8,0::C:4
led 9 0,1::C:4
led 10 1,1::C:4
led 11 2,1::C:4
led 12 3,1::C:4
led 13 4,1::C:4
led 14 5,1::C:4
led 15 6,1::C:4
led 16 7,1::C:4
led 17 8,1::C:4

# color
color 8 240,0,255

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 3 900 1700 0 0
aux 2 13 1 1700 2100 0 0
aux 3 26 1 1300 1700 0 0

# adjrange
adjrange 0 0 3 900 2100 12 2 0 0

# master
set dyn_notch_count = 1
set dyn_notch_q = 120
set dyn_notch_min_hz = 200
set dyn_notch_max_hz = 500
set acc_calibration = -6,34,-87,1
set align_mag = CW90
set mag_align_yaw = 900
set mag_calibration = -443,72,89
set baro_hardware = NONE
set serialrx_provider = CRSF
set dshot_idle_value = 800
set dshot_bidir = ON
set vbat_max_cell_voltage = 440
set yaw_motors_reversed = ON
set small_angle = 180
set gps_sbas_mode = EGNOS
set gps_ublox_use_galileo = ON
set pid_process_denom = 2
set osd_warn_bitmask = 148445
set osd_vbat_pos = 701
set osd_rssi_pos = 2459
set osd_link_quality_pos = 2748
set osd_rssi_dbm_pos = 547
set osd_rsnr_pos = 2716
set osd_flymode_pos = 2464
set osd_anti_gravity_pos = 1633
set osd_throttle_pos = 2735
set osd_crosshairs_pos = 2460
set osd_ah_sbar_pos = 303
set osd_ah_pos = 206
set osd_current_pos = 2401
set osd_mah_drawn_pos = 0
set osd_craft_name_pos = 484
set osd_gps_speed_pos = 2684
set osd_gps_sats_pos = 136
set osd_home_dir_pos = 1668
set osd_home_dist_pos = 2627
set osd_flight_dist_pos = 416
set osd_altitude_pos = 3716
set osd_power_pos = 388
set osd_warnings_pos = 14586
set osd_avg_cell_voltage_pos = 3748
set osd_pit_ang_pos = 2740
set osd_rol_ang_pos = 456
set osd_battery_usage_pos = 9
set osd_disarmed_pos = 2336
set osd_esc_tmp_pos = 416
set osd_esc_rpm_pos = 288
set osd_rate_profile_name_pos = 2496
set osd_pid_profile_name_pos = 2528
set osd_sys_lq_pos = 664
set osd_stat_bitmask = 42467132
set osd_displayport_device = MSP
set debug_mode = GYRO_SCALED
set vcd_video_system = HD
set craft_name = Bumblebee HD V3

profile 0

# profile 0
set dterm_lpf1_dyn_min_hz = 77
set dterm_lpf1_dyn_max_hz = 187
set dterm_lpf2_static_hz = 165
set vbat_sag_compensation = 100
set iterm_relax_cutoff = 10
set throttle_boost = 0
set p_pitch = 66
set i_pitch = 99
set d_pitch = 47
set f_pitch = 110
set p_roll = 63
set i_roll = 94
set d_roll = 45
set f_roll = 105
set p_yaw = 53
set i_yaw = 99
set f_yaw = 100
set d_min_roll = 0
set d_min_pitch = 0
set feedforward_transition = 40
set dyn_idle_min_rpm = 15
set simplified_pids_mode = OFF
set simplified_dterm_filter = OFF
set tpa_mode = PD
set tpa_rate = 45
set tpa_breakpoint = 1550

profile 1

profile 2

profile 3

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set thr_mid = 35
set thr_expo = 70
set roll_rc_rate = 6
set pitch_rc_rate = 6
set yaw_rc_rate = 6
set roll_expo = 60
set pitch_expo = 60
set yaw_expo = 60
set roll_srate = 50
set pitch_srate = 50
set yaw_srate = 50
set throttle_limit_type = SCALE
set throttle_limit_percent = 80

rateprofile 1

# rateprofile 1
set thr_mid = 35
set thr_expo = 50
set roll_rc_rate = 6
set pitch_rc_rate = 6
set yaw_rc_rate = 6
set roll_expo = 60
set pitch_expo = 60
set yaw_expo = 60
set roll_srate = 50
set pitch_srate = 50
set yaw_srate = 50
set throttle_limit_type = SCALE
set throttle_limit_percent = 80

rateprofile 2

# rateprofile 2
set thr_mid = 35
set thr_expo = 65
set roll_rc_rate = 2
set pitch_rc_rate = 2
set yaw_rc_rate = 6
set roll_expo = 54
set pitch_expo = 54
set yaw_expo = 54
set throttle_limit_type = SCALE

rateprofile 3

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#
```

haslinghuis, 2023-01-04:
Try adding one of these custom defines while flashing. Please let us know which one so we can add the correct define to configuration.
```
FLASH_M25P16
FLASH_W25N01G
FLASH_W25M
FLASH_W25M512
FLASH_W25M02G
FLASH_W25Q128FV
```

MaryJane136, 2023-01-05:
Hi. You wrote on Discord you already added the missing define I found in the source file. 
https://github.com/betaflight/unified-targets/pull/788
But it should be configurable for non-developers in addition.

closed 2023-01-05 by MaryJane136