betaflight/betaflight-configurator #3231: Motor Direction Wizzard only shows 4 Motors

jappyjan, 2023-01-11:
### Describe the bug

I have a Y7 Copter.
I can control each of the 6 motors through the sliders on the Motors Tab.
When opening the Motor Direction Wizzard (Individual and "Wizard" option both same problem) only 4 of the 6 Motors are configurable.

### To Reproduce

Build a Y6 Copter.
Connect the 6 Motors to the first 6 ESC Outputs.
Connect the Copter to Betaflight (as my usb is unreachable right now, i use the ELRS Wifi for this, maybe this causes the problem?)
Go to Motors Tab.
Test all motors are controllable and show up by using the Motor Sliders.
Open the Motor Direction Wizzard and try to change the Motor Directions.

### Expected behavior

I can see and modify all 6 Motor Directions.

### Configurator version

10.8.0

### Add any other context about the problem that you think might be relevant here

The "reorder motors" dialog works flawlessly though

haslinghuis, 2023-01-11:
Which mixer model have you set up? The mixer models decides how many motors can be used at the time depending on support from the FC itself.

jappyjan, 2023-01-11:
its the Y6 mixer model

haslinghuis, 2023-01-11:
What is the output of `dshot_telemetry_info`?

[image]

jappyjan, 2023-01-11:
if you mean the error numbers which are red on your screenshot above the motor sliders:
all at 0%

if you mean the graphs at the top, looking normal like on any other of my quads, not too much, not too little noise, all in sync with motor speed.

haslinghuis, 2023-01-11:
Configurator gets the number of available motor output from firmware.

Lehtusbphree2flyFPV, 2023-06-01:
THE Motor direction  not working right they wont reverse the direction it spins

Prajwal-Koirala, 2023-12-27:
**Here's a summary of the GitHub issue #3231, incorporating insights from the provided text:**

**Issue Title:** Motor Direction Wizard only shows 4 Motors

**Status:** Open

**Summary:**

- User jappyjan reports that the Motor Direction Wizard in Betaflight Configurator only displays 4 out of 6 motors for their Y7 Copter, even though all motors are controllable via the sliders on the Motors tab.
- The issue was initially labeled as a bug, but later reclassified as a feature request to extend the wizard's functionality to non-standard model types.

**Key Discussion Points:**

- **Mixer Model:** The chosen mixer model influences the number of motors the configurator recognizes.
- **Firmware and Configurator Versions:** Compatibility between firmware and configurator versions is important.
- **Telemetry Information:** The `dshot_telemetry_info` command can provide insights into motor health and communication.
- **Available Motor Outputs:** The `resource show all` command reveals the number of motor outputs recognized by the firmware.
- **ELRS WiFi Connectivity:** The possibility of ELRS WiFi being a contributing factor was raised but not conclusively determined.

**Current Status and Next Steps:**

- The issue is still open, awaiting further input or a potential feature implementation.
- Contributors have suggested using Discord for more immediate assistance and bug reporting.
- Users experiencing similar issues are encouraged to provide detailed information about their setups, including:
    - Flight controller model
    - Firmware version
    - Configurator version
    - Relevant output from commands like `dshot_telemetry_info` and `resource show all`

**Additional Notes:**

- The "Reorder Motors" dialog reportedly works correctly for the affected user.
- Another user, Lehtusbphree2flyFPV, has also encountered motor direction issues, but their case might be distinct.

**Recommendations:**

- Stay updated on the issue thread for any progress or workarounds.
- Consider joining the Betaflight Discord server for real-time assistance.
- If you encounter the same problem, provide comprehensive details about your setup to aid in troubleshooting.

closed 2024-03-23 by jappyjan