betaflight/betaflight-configurator #3240: Motor test mode - slider not working

blink1n, 2023-01-13:
### Describe the bug

Motor test mode sliders not working when mode activated 

### To Reproduce

betaflight-configurator_10.9.0-RC4_win64-portable.zip

Connect working drone with any BF with battery
Go to motor tab
enable test mode
Try to move slider

### Expected behavior

motor spin when slider is enabled and moved.

### Configurator version

2023-01-13 @16:24:31 -- OS: Windows  2023-01-13 @16:24:31 -- Configurator: 10.9.0-RC4 (norevision)

### Add any other context about the problem that you think might be relevant here

tested with 4.3 and 4.4 rc4

haslinghuis, 2023-01-13:
@blink1n this is a known issue in RC4 and is fixed in development.

Please use: https://github.com/betaflight/betaflight-configurator-nightlies/releases

blink1n, 2023-01-13:
Thanks we are aware of issue. #3227

closed 2023-01-13 by blink1n