betaflight/betaflight-configurator #3242: Option for automatic backup

Combinacijus, 2023-01-14:
### Is your feature request related to a problem? Please describe

When experimenting with configuration there's many tests to check how different configs and after finishing setup one might forget to do "Save Backup" or "diff all" to record latest setup or backup after minor tweak. And if something happens latest config is lost

### Describe the solution you'd like

I'd like to have an option for automatic backups in some folder after every change (I guess it's when Save button is pressed). File should be uniquely named with a craft name, pilot name, board name, date or similar

Having backups would help in 2 cases:
1. You can recover lost config if FC dies etc
2. You had previous working config and you want to go back (like in git history)

And generally it would save time while configuring because you won't have to manually backup
[image]

### Describe alternatives you've considered

No alternatives

Prajwal-Koirala, 2023-12-27:
**+1 for Automatic Backups in Betaflight Configurator**

**I wholeheartedly support the implementation of automatic backups in Betaflight Configurator.** Here's why:

**Peace of Mind During Configuration:**

- ️ **Safeguards against accidental changes or FC failures.**
- ⏱️ **Saves time by eliminating manual backup steps.**
-  **Encourages experimentation and fine-tuning without fear of losing progress.**

**Benefits in Real-World Scenarios:**

-  **Recover lost configurations** due to unexpected events.
-  **Revert to previous working versions** for reliability and troubleshooting.
-  **Track configuration history** like a mini Git for your quad.

**Vision for a User-Friendly Implementation:**

- **Automatic backup after every save**, ensuring seamless protection.
- **Uniquely named files** based on craft name, pilot name, board name, and date for easy identification.
- **Customizable backup folder** to match user preferences.
- **Optional toggle** for users who prefer manual backups.

**Incorporating Visual Feedback:**

- **Clear visual indicators** when backups are created and restored, enhancing user confidence.
- **Consider a "backup history" tab** to list and manage previous versions, similar to Git log visualization.

**I'm confident that automatic backups would be a valuable addition to Betaflight Configurator, streamlining workflows and providing a safety net for pilots. Let's make it happen!**

**Image of a quadcopter with a green checkmark, symbolizing peace of mind with automatic backups: [https://user-images.githubusercontent.com/16375702/212502315-39b0ac49-0a2d-4fe2-840b-64c6e9186638.png](https://user-images.githubusercontent.com/16375702/212502315-39b0ac49-0a2d-4fe2-840b-64c6e9186638.png)**

haslinghuis, 2024-05-05:
Latest configurator now gives the option to make a diff backup upon flashing.

closed 2024-05-05 by haslinghuis