betaflight/betaflight-configurator #3243: Configurator 10.9.0-RC4 on Ubuntu won't make it past loading ports

Schlaegel, 2023-01-15:
### Describe the bug

On release Configurator 10.8.0 (deb) everything works as expected.
On RC4 Configurator 10.9.0-RC4 (deb) the port drop down stays at Loading...
On RC4 Configurator 10.9.0-RC4 (linux64 portable) the port drop down stays at Loading...

I can switch back and forth between 10.8.0 and 10.9.0-RC4 and always get the same results.
I can reboot the computer and start 10.9.0-RC4 and get the same result ("Loading...")
I can reboot the computer and start 10.8.0 and get the same result (It works)

So as is, 10.9.0-RC4 won't run at all on my Ubuntu box.

```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.10
Release:	22.10
Codename:	kinetic
```

```
$ ls -l betaflight-configurator_10*
-rw-rw-r-- 1 aaron aaron 105173876 Jun 13  2022 betaflight-configurator_10.8.0_amd64.deb
-rw-rw-r-- 1 aaron aaron 109948728 Jan 14 21:40 betaflight-configurator_10.9.0-RC4_amd64.deb
-rw-rw-r-- 1 aaron aaron 165448967 Jan 14 21:55 betaflight-configurator_10.9.0-RC4_linux64-portable.zip

```

### To Reproduce

On ubuntu 22.10, install the deb and run the configurator.
OR
On ubuntu 22.10, extract the portable zip and run the configurator.

### Expected behavior

The configurator loads a list of ports that have FC connected to them.

### Configurator version

10.9.0-RC4

### Flight controller configuration

```txt
NA
```

haslinghuis, 2023-01-15:
Betaflight only supports LTS releases. Ubuntu 22.04 works for development.

Schlaegel, 2023-01-15:
Is this a won't fix then? If so, it would be helpful to say something on the download page about not supporting Ubuntu 22.10. Where does it say the distro releases that are supported?

If you want to look into this and get it working, I'll work with you.

haslinghuis, 2023-01-15:
Sure I'll fix this. Doing this kind of work for 36 years now. Just have to find my USB stick and install it so see what's going on :)

haslinghuis, 2023-01-15:
There is nothing to fix. It just works here.

https://github.com/betaflight/betaflight-configurator#linux-users

[image]

closed 2023-01-29 by haslinghuis