betaflight/betaflight-configurator #3244: Motor tab misbehaviour

N-boobook, 2023-01-15:
### Describe the bug

I took a default setting flight controller and a new installation of betaflight configurator 10.9.0 RC4. Reordering the motors and direction was successful but the configurator gave no indication that it had happened.

### To Reproduce

Try and reorder the motors or change direction on a board_name MAMBAF722_2022B with RC4 firmware and then save. Nothing appears to happen. The changes occur despite no obvious feedback.

### Expected behavior

I would expect the save button to respond.

### Configurator version

10.9.0-RC4

### Add any other context about the problem that you think might be relevant here

I am new to this but keen on working out how to add to the effort. I appreciate using this software.

haslinghuis, 2023-01-15:
Please try: https://github.com/betaflight/betaflight-configurator-nightlies/releases

haslinghuis, 2023-01-29:
Fixed in #3247

closed 2023-01-29 by haslinghuis