betaflight/betaflight-configurator #3256: Launch Control missing

Spit29, 2023-01-17:
### Describe the bug

Hi everyone, I was trying to setup launch control on betaflight 4.4.0 RC5 but launch control seems to be missing from the CLI commands.. 

set launch_control_mode = PITCHONLY
set launch_trigger_allow_reset = ON
set launch_trigger_throttle_percent = 50
set launch_angle_limit = 75
set osd_warn_launch_control = ON

### To Reproduce

These commands no longer work in CLI
set launch_control_mode = PITCHONLY
set launch_trigger_allow_reset = ON
set launch_trigger_throttle_percent = 50
set launch_angle_limit = 75
set osd_warn_launch_control = ON

### Expected behavior

"Set Launch" shows nothing in CLI

### Configurator version

10.9.0 RC5

### Flight controller configuration

```txt
# 

# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.4.0 Jan 17 2023 / 04:30:33 (a7cf8f0c6) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name SPEEDYBEEF7V3
manufacturer_id SPBE

# feature
feature GPS

# serial
serial 0 131073 115200 57600 0 115200
serial 5 2 115200 57600 0 115200

# beacon
beacon RX_LOST
beacon RX_SET

# aux
aux 0 0 0 1850 2100 0 0
aux 1 1 3 1675 2100 0 0
aux 2 13 1 1900 2100 0 0
aux 3 35 2 1875 2100 0 0
aux 4 46 1 1350 1675 0 0
aux 5 49 2 1400 1600 0 0

# master
set gyro_lpf1_static_hz = 0
set gyro_lpf2_type = PT2
set gyro_lpf2_static_hz = 135
set dyn_notch_count = 1
set dyn_notch_q = 500
set gyro_lpf1_dyn_min_hz = 0
set acc_calibration = 26,-18,61,1
set baro_hardware = NONE
set rc_smoothing_auto_factor = 250
set rc_smoothing_auto_factor_throttle = 250
set rc_smoothing_setpoint_cutoff = 12
set rc_smoothing_feedforward_cutoff = 12
set rc_smoothing_throttle_cutoff = 20
set blackbox_sample_rate = 1/1
set blackbox_device = NONE
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_output_reordering = 1,3,0,2,4,5,6,7
set failsafe_procedure = GPS-RESCUE
set bat_capacity = 1400
set beeper_dshot_beacon_tone = 3
set small_angle = 180
set gps_rescue_allow_arming_without_fix = ON
set deadband = 1
set yaw_deadband = 1
set simplified_gyro_filter = OFF
set osd_units = IMPERIAL
set osd_rssi_alarm = 40
set osd_rssi_dbm_alarm = -98
set osd_cap_alarm = 1200
set osd_alt_alarm = 400
set osd_vbat_pos = 3622
set osd_rssi_pos = 410
set osd_link_quality_pos = 3557
set osd_rssi_dbm_pos = 2557
set osd_tim_1_pos = 53
set osd_tim_2_pos = 3493
set osd_flymode_pos = 390
set osd_throttle_pos = 384
set osd_vtx_channel_pos = 16
set osd_crosshairs_pos = 236
set osd_ah_pos = 199
set osd_current_pos = 2546
set osd_mah_drawn_pos = 2622
set osd_craft_name_pos = 28
set osd_gps_speed_pos = 3174
set osd_gps_lon_pos = 18518
set osd_gps_lat_pos = 18486
set osd_gps_sats_pos = 2097
set osd_home_dir_pos = 2236
set osd_home_dist_pos = 2204
set osd_compass_bar_pos = 265
set osd_altitude_pos = 19526
set osd_pid_roll_pos = 135
set osd_pid_pitch_pos = 167
set osd_pid_yaw_pos = 199
set osd_debug_pos = 0
set osd_power_pos = 320
set osd_pidrate_profile_pos = 344
set osd_warnings_pos = 2191
set osd_avg_cell_voltage_pos = 2611
set osd_pit_ang_pos = 256
set osd_rol_ang_pos = 288
set osd_battery_usage_pos = 2073
set osd_disarmed_pos = 75
set osd_nheading_pos = 310
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_displayport_device = MSP
set osd_canvas_width = 60
set osd_canvas_height = 22
set debug_mode = GYRO_SCALED
set vcd_video_system = HD

profile 0

# profile 0
set dterm_lpf1_dyn_min_hz = 97
set dterm_lpf1_dyn_max_hz = 195
set dterm_lpf1_static_hz = 97
set dterm_lpf2_static_hz = 195
set iterm_relax_cutoff = 10
set throttle_boost = 0
set p_pitch = 59
set i_pitch = 144
set d_pitch = 37
set f_pitch = 202
set p_roll = 54
set i_roll = 130
set d_roll = 33
set f_roll = 184
set p_yaw = 54
set i_yaw = 130
set f_yaw = 184
set d_min_roll = 33
set d_min_pitch = 37
set motor_output_limit = 80
set feedforward_averaging = 2_POINT
set feedforward_smooth_factor = 70
set feedforward_jitter_factor = 13
set dyn_idle_min_rpm = 30
set simplified_master_multiplier = 110
set simplified_i_gain = 135
set simplified_pi_gain = 110
set simplified_dmax_gain = 0
set simplified_feedforward_gain = 140
set simplified_pitch_pi_gain = 105
set simplified_dterm_filter_multiplier = 130

rateprofile 0

# rateprofile 0
set roll_rc_rate = 1
set pitch_rc_rate = 1
set yaw_rc_rate = 1
set roll_expo = 55
set pitch_expo = 55
set yaw_expo = 55
set roll_srate = 100
set pitch_srate = 100
set yaw_srate = 70

# end the command batch
batch end

#
```

haslinghuis, 2023-01-17:
Please add the features you require while flashing your board.

https://betaflight.com/docs/configurator/firmware-flasher-tab#custom-defines

Spit29, 2023-01-17:
Got it! Thank you

Spit29, 2023-01-17:
> Please add the features you require while flashing your board.
> 
> https://betaflight.com/docs/configurator/firmware-flasher-tab#custom-defines

Just 1 thing missing, will it display in the OSD if launch control is on? This option doesn't seem to work osd_warn_launch_control = ON

haslinghuis, 2023-01-17:
Got replace by @limonspb 
```
# get osd_warn_bitmask
osd_warn_bitmask = 8191
Allowed range: 0 - 4294967295
```
See: https://github.com/betaflight/betaflight/pull/11671

You can set them up more easy using osd tab:

[image]

closed 2023-01-17 by Spit29