betaflight/betaflight-configurator #3260: Betaflight 10.9 RC5 won't open on OSX 10.13.6

ImmeFPV, 2023-01-18:
### Describe the bug

After allowing the unverified developer in the OSX safety menu, beta flight starts but then nothing happens. The configurator window doesn't show. 

Tried other RCs but it was the same. Won't start.

It worked perfectly in 10.8 and before...

### To Reproduce

- OSX 10.13.6
- Download betaflight-configurator_10.9.0-RC5_macOS.dmg
- open
- drag to apps folder
- open BF
- go to OSX safety and allow
- ...

### Expected behavior

Open the configurator main menu window, where I can flash FW or just connect to the FC

### Configurator version

10.9.

haslinghuis, 2023-01-19:
https://github.com/nwjs/nw.js/issues/7946

haslinghuis, 2023-01-25:
To be able to run 10.9 follow this workaround:

- install betaflight configurator 10.8
- delete the framework folder in 10.9
- copy the framework folder from 10.8 into 10.9
- remove betaflight configurator 10.8

[image]

roligmue, 2023-02-01:
Hi,

This workaround does work with the newly available BF Configurator 10.9.0 too.

However, you have to also delete and copy over the MacOS folder too. The graphic that was shared shows both directories, but the written instructions do not include the MacOS directory. 

The following worked for me.

To be able to run 10.9 follow this workaround:

install betaflight configurator 10.8
delete the framework & the MacOS folder in 10.9
copy the framework & MacOS folder from 10.8 into 10.9
remove betaflight configurator 10.8

ImmeFPV, 2023-02-10:
Sorry, just got the chance to try your solution, - WORKS!!!!

Thank you both so much!!!!!! <3

roligmue, 2023-02-10:
Good deal!

Mod-Quad, 2023-02-26:
Working for me as well - hope no unintended side-effects that might compromise safety.

delfer, 2023-03-12:
Hello! I'm using my MacBook Pro (13-inch, Early 2011) for the DIY to prevent blowing new one. And the latest available macOS version for this device is macOS High Sierra 10.13.16.
[Betaflight Configurator 10.9.0](https://github.com/betaflight/betaflight-configurator/releases/tag/10.9.0) wan't work with macOS High Sierra 10.13.16. The application's window does not appear when launching.
[Betaflight Configurator Nightly](https://github.com/betaflight/betaflight-configurator-nightlies/releases) [v20230309.403](https://github.com/betaflight/betaflight-configurator-nightlies/releases/tag/v20230309.403) also not working.
```

Last login: Sun Mar 12 14:02:39 on ttys001
/Applications/Betaflight\ Configurator.app/Contents/MacOS/nwjs ; exit;
MacBook-Pro-user:~ delfer$ /Applications/Betaflight\ Configurator.app/Contents/MacOS/nwjs ; exit;
dyld: lazy symbol binding failed: Symbol not found: ____chkstk_darwin
  Referenced from: /Applications/Betaflight Configurator.app/Contents/Frameworks/nwjs Framework.framework/Versions/109.0.5414.87/libnode.dylib (which was built for Mac OS X 10.15)
  Expected in: /usr/lib/libSystem.B.dylib

dyld: Symbol not found: ____chkstk_darwin
  Referenced from: /Applications/Betaflight Configurator.app/Contents/Frameworks/nwjs Framework.framework/Versions/109.0.5414.87/libnode.dylib (which was built for Mac OS X 10.15)
  Expected in: /usr/lib/libSystem.B.dylib

Received signal 6
 [0x00010ff30db2]
 [0x00010fe83f83]
 [0x00010ff30d01]
 [0x7fff6a805f5a]
 [0x00010ba9750b]
 [0x00010baa1bf5]
 [0x00010ba75482]
 [0x00010ba755a9]
 [0x7fff6a4f7292]
 [0x000108f18c10]
 [0x000107635808]
 [0x00010777ab67]
 [0x0001157ef59d]
 [0x0001157efc9b]
 [0x00010f9c9bae]
 [0x00010f9ca743]
 [0x00010f9c8af4]
 [0x00010f9c930b]
 [0x00010bb1243e]
 [0x000103b4e9b9]
 [0x7fff6a4f7015]
 [0x000000000015]
[end of stack trace]
[0312/150920.997337:WARNING:process_memory_mac.cc(93)] mach_vm_read(0x7ffeec0b1000, 0x2000): (os/kern) invalid address (1)

```
At the same time [Betaflight - Configurator Chrome extension](https://chrome.google.com/webstore/detail/betaflight-configurator/kdaghagfopacdngbohiknlhcocjccjao) working good, but the latest available version 10.4.1 is out of date.
@haslinghuis is it possible to downgrade NW.js back to 0.64.0 (revert https://github.com/betaflight/betaflight-configurator/commit/c8d0490b2ffa66615ea37da0e9a7282af8d01077)? According this comment https://github.com/nwjs/nw.js/issues/7946#issuecomment-1415392164

> NW.js 0.64.0 is the lastest version I got running under macOS 10.13.6 (High Sierra) on my old Mac mini mid-2011

delfer, 2023-03-12:
Everyone can use this zip's I've uploaded from 10.8.0 image
https://github.com/delfer/betaflight-configurator_10.8.0_macOS/releases/download/v10.8.0/MacOS.zip
https://github.com/delfer/betaflight-configurator_10.8.0_macOS/releases/download/v10.8.0/Frameworks.zip

byavallone, 2023-04-20:
The workaround works for now but is there a chance to get something official for the next version?

haslinghuis, 2023-04-20:
PWA version of configurator might fix this in the future.

closed 2023-04-20 by haslinghuis