betaflight/betaflight-configurator #3261: Please add Tooltip or wiki reference for "Other options" page

rally-dimi, 2023-01-19:
### Is your feature request related to a problem? Please describe

I'm new to FPV so I don't know every feature of my hardware - and it is hard to say what some of those "Other options" mean. 
I was searching for wiki for some, but found just code. 

[image]

### Describe the solution you'd like

Please add tooltip to each selected option, so when you hover over it with mouse, it would put description. 
Or create a link near the selections that would redirect you to a wiki page with options.

### Describe alternatives you've considered

Just selecting minimal options without bothering.

### Other information

No other info.

haslinghuis, 2023-01-19:
What do you see In the upper right corner of configurator in the flasher tab?

rally-dimi, 2023-01-20:
Wow, it could be even tinier.  :smile: 
Thank you! @haslinghuis 
Maybe one note - can you add comment - `see description of features in **wiki**` ?
For example - in the same line as the `Other options` label ?

rally-dimi, 2023-01-20:
Also those features on wiki could be a bit more descriptive - not sure what `PINIO` means. Or `Enable VTX` - like without it wouldn't be able to transmit video ?

closed 2023-01-20 by rally-dimi