betaflight/betaflight-configurator #3266: Not possible to save the feature "Blackbox logging" in the cli-settings

pape-engineering, 2023-01-22:
### Describe the bug

It is not possible to save the feature "Blackbox" logging in a backup (cli-settings).

### To Reproduce

Enable Blackbox logging under features, save a backup and restore from a backup. Blackbox logging is disabled.

### Expected behavior

It should be saved like other features

### Configurator version

10.9.0-RC5

### Add any other context about the problem that you think might be relevant here

ID for the uploaded support-data: d48b05b2-2fad-49a0-84bd-c7775d295660

haslinghuis, 2023-01-23:
Seems the `BLACKBOX` feature does not exist for a really long time. So we need to remove the feature from Configurator.

Found two other features not being used anymore: `CURRENT_METER` and `VBAT`

closed 2023-01-23 by haslinghuis

pape-engineering, 2023-01-23:
Yeah it was not woking since a long time... So it's a very good idea to remove all deprecated options. Then it will not confusing anymore.

PR #3268 (2023-01-23): Remove deprecated features
Fixes: #3266