betaflight/betaflight-configurator #3267: Not possible to save the osd-element numerical climb rate (vario) in the cli-settings.

pape-engineering, 2023-01-22:
### Describe the bug

It is not possible so save the position of the osd-element "numerical climb rate" (vario) in the cli-settings. Variable not listed in the cli-settings.

### To Reproduce

Enable the element for numerical climb rate (vario) in osd, move it around, save the position, create a backup and restore from a backup. The osd-element is missing.

### Expected behavior

It should saved like other osd-elements.

### Configurator version

10.9.0-RC5

### Add any other context about the problem that you think might be relevant here

ID for the uploaded support-data: d48b05b2-2fad-49a0-84bd-c7775d295660

haslinghuis, 2023-01-23:
Please add `VARIO` to custom defines. Needs option `GPS`. Baro is present on the board.

pape-engineering, 2023-01-23:
I guess you mean to put VARIO to custom defines for cloud building. It's a very helpful hint because I was not using this custom define. Just an idea.... it would be helpful to see an overview of all custom defines in the wiki or in a kind of drop-down list like with the options "Flash" "GPS" "MAG" and so on. Thank you very much. I will give it a try.

haslinghuis, 2023-01-23:
Please check the wiki button (needs nightly build)

https://betaflight.com/docs/configurator/firmware-flasher-tab#custom-defines

closed 2023-01-29 by haslinghuis