betaflight/betaflight-configurator #3282: "Failed to open serial port" in Linux

RahulGoyal-tech, 2023-01-27:
### Describe the bug

I have connected the flight controller to pc and it is being automatically detected by betaflight but whenever i try to connect, it shows failed to open serial port. Please suggest what should i do because my kernel is 5.x.x version so there is no installation available for cp210x

### To Reproduce

happening in my system

### Expected behavior

connect successfully to fc

### Configurator version

10.8.0

haslinghuis, 2023-01-27:
Please check https://betaflight.com/docs/wiki/archive/Installing-Betaflight#platform-specific-linux

RahulGoyal-tech, 2023-01-28:
Thank you so much. It worked perfectly 😊

closed 2023-01-28 by haslinghuis

TheForsakenSpirit, 2024-01-24:
@haslinghuis can you update link or add solution description?

haslinghuis, 2024-01-24:
Link has been updated.

nerdCopter, 2024-01-24:
> @haslinghuis can you update link or add solution description?

@TheForsakenSpirit , this is fairly concise as well: https://hackmd.io/@nerdCopter/H1dtIuUSn

passalini, 2024-05-02:
after pluging usb run `sudo chmod 0777 /dev/ttyACM0`

adii-n, 2024-05-11:
`sudo chmod 777 /dev/tty/ACM0` works perfectly for me to make the com port recognizable. The only issue is that I have to run the command every time betaflight restarts; is there any way to make the change permanent?

nerdCopter, 2024-05-13:
0777 is a workaround, the proper solution is udev rules and user groups.  see official docs or the link i posted above.

kiranze, 2024-08-22:
> One more solution (without building) after pluguing usb run: `sudo chmod 0777 /dev/ttyACM0`

Thanks man, worked immediately

Juribiyan, 2024-09-13:
For udev rules and usergroups follow the [instructions provided by the ExpressLRS Configurator].(https://github.com/ExpressLRS/ExpressLRS-Configurator?tab=readme-ov-file#linux). It worked for me.