betaflight/betaflight-configurator #3285: Betaflight Configurator 10.9.0 APK: Same issue than https://github.com/betaflight/betaflight-configurator/issues/3048

Canaill51, 2023-01-28:
### Describe the bug

Exact same issue with latest 10.9.0 GUI APK, than one reported in https://github.com/betaflight/betaflight-configurator/issues/3048

### To Reproduce

Same than https://github.com/betaflight/betaflight-configurator/issues/3048

### Expected behavior

Same than https://github.com/betaflight/betaflight-configurator/issues/3048

### Configurator version

10.9.0 (697943b)

haslinghuis, 2023-01-28:
@McGiverGim - we should document update process for next release.

@Canaill51 Please use `10.10`: https://github.com/betaflight/betaflight-configurator-nightlies/releases/tag/v20230128.362

McGiverGim, 2023-01-28:
And use a real certificate... Except if we go to pwa.

Canaill51, 2023-01-29:
@haslinghuis just removed 10.9.0 GUI from my Android smartphone and installed 10.10 from your above link:
No message no longer occurs.

Could 10.9.0 get this fix?

Canaill51, 2023-02-20:
@haslinghuis, any update since this ?
As confirmed, 10.10 is not showing up this messages.
Which apk should we use on Android, to configure BF ?
Is it planned to fix 10.9 ?

haslinghuis, 2023-02-20:
We are working on it: https://github.com/betaflight/betaflight-configurator/pull/3349

closed 2023-03-31 by github-actions[bot]