betaflight/betaflight-configurator #3289: GPS Matek and T-Motor HD F7 configuration problem

FedericoMarcoDariol, 2023-01-29:
### Describe the bug

On the Configuration page I enable the GPS item, enable the necessary ports, save and after saving it is not enabled although on the Ports page I have enabled everything correctly.

### To Reproduce

PORTS page I enable UART 4 for GPS
CONFIGURATION page, GPS item enabled
SAVE and RESTART
I go back to the CONFIGURATION page and the GPS item is disabled

### Expected behavior

I would like the GPS port to remain enabled

### Configurator version

Betaflight 4.4.0

### Flight controller configuration

```txt
# 

# diff all

# version
# Betaflight / STM32F7X2 (S7X2) 4.4.0 Jan 29 2023 / 06:51:38 (4605309d8) MSP API: 1.45

# config: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name TMOTORF7
manufacturer_id TMTR
mcu_id 002800224847500220343432
signature 

# name: IronMan MK7

# feature
feature TELEMETRY

# serial
serial 0 32 115200 57600 0 115200
serial 1 131073 115200 57600 0 115200
serial 3 2 115200 57600 0 115200
serial 4 64 115200 57600 0 115200

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# aux
aux 0 0 0 1900 2100 0 0
aux 1 1 1 1350 1550 0 0
aux 2 2 1 900 1050 0 0
aux 3 13 2 1900 2100 0 0
aux 4 36 3 1900 2100 0 0

# master
set dyn_notch_count = 1
set dyn_notch_q = 500
set acc_calibration = 1,32,49,1
set align_mag = CW90FLIP
set mag_align_pitch = 1800
set mag_align_yaw = 900
set rssi_channel = 12
set serialrx_provider = CRSF
set blackbox_sample_rate = 1/8
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set failsafe_procedure = GPS-RESCUE
set ibata_scale = 200
set yaw_motors_reversed = ON
set small_angle = 180
set osd_cap_alarm = 4500
set osd_alt_alarm = 1000
set osd_vbat_pos = 2496
set osd_rssi_pos = 2112
set osd_link_quality_pos = 64
set osd_flymode_pos = 2080
set osd_current_pos = 384
set osd_mah_drawn_pos = 2432
set osd_craft_name_pos = 2508
set osd_gps_speed_pos = 2336
set osd_gps_lon_pos = 18448
set osd_gps_lat_pos = 18435
set osd_gps_sats_pos = 2304
set osd_home_dir_pos = 2094
set osd_home_dist_pos = 2272
set osd_flight_dist_pos = 2208
set osd_altitude_pos = 2240
set osd_avg_cell_voltage_pos = 2464
set osd_disarmed_pos = 2346
set osd_displayport_device = MSP
set vcd_video_system = HD
set craft_name = IronMan MK7

profile 0

profile 1

profile 2

profile 3

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#
```

McGiverGim, 2023-01-29:
You need to enable the GPS feature in the cloud build options when flashing.

FedericoMarcoDariol, 2023-01-29:
> You need to enable the GPS feature in the cloud build options when flashing.

I use the classic configuration when flashing

haslinghuis, 2023-01-29:
Classic (now core only) contains only driver support and bare minimum of features for support.

closed 2023-02-01 by haslinghuis