betaflight/betaflight-configurator #3297: Add an option to cache presets for offline usage

glennvenghaus, 2023-01-30:
### Is your feature request related to a problem? Please describe

The preset feature is very powerfull and due to its modularity can be used to program a quad in the field with prepared presets at home (assuming own github repo) . Currently this requires an active internet connection which is not always available in the field.

### Describe the solution you'd like

A basic caching feature that would keep previously loaded preset repos in local cache and only update it when internet connection is available. 
A user option to enable caching for selected repos or globaly could be handy, as well as a force update or flush cache button in case you want to be 100% sure you have the latest versions.

A personal current use case is, in the field loading a highly customisable presets for parameters not part of profiles and based on the Quad/Lipo  combination used. These set  RTH hover position, battery alert levels , throttle mid etc. I can easily switch between a light weight lipo or a heavy Lion for long range by just clicking a preset. 
Currently i can only do this when internet is available.

### Describe alternatives you've considered

Alternative would be to have more parameters part of profiles in betaflight that , like PID profiles , can be switched with the OSD menu.  Gps_rescue, battery alerts as exampled in my use case are not part of any profile atm.

Benky, 2023-04-18:
As a workaround (I use in the field) you can clone presets Github repository (`https://github.com/betaflight/firmware-presets`) locally and create new Preset source in Betaflight configurator with URL like `file:///opt/repos/github/firmware-presets` (basically a path to a dir where you've cloned presets repository to prefixed with `file:/`)

glennvenghaus, 2023-04-18:
Thanks !!!!! 
I did not know about the file:/ option. That would indeed help for a laptop based offline solution in the field .

haslinghuis, 2024-05-05:
Presets are cloud based. Having a local repo like @Benky suggested solves the issue.

closed 2024-05-05 by haslinghuis