betaflight/betaflight-configurator #3304: Building

smart2pet, 2023-02-02:
### Describe the bug

I can't build this project. Probably it's because the network in China, and it uses google resources so I cant reach it. 

### To Reproduce

Use a computer in China, and simply use `npm install`, and you'll got the error.

### Expected behavior

We can build it in China. 

### Configurator version

10.9.0

### Add any other context about the problem that you think might be relevant here

Output log:
```
Output:
node:events:491
      throw er; // Unhandled 'error' event
      ^
ErrorClass [RequestError]: connect ETIMEDOUT 142.251.42.225:443
    at ClientRequest.<anonymous> (D:\zry\dev\nodejs\betaflight-configurator\node_modules\got\index.js:73:21)
    at Object.onceWrapper (node:events:628:26)
    at ClientRequest.emit (node:events:513:28)
    at TLSSocket.socketErrorListener (node:_http_client:494:9)
    at TLSSocket.emit (node:events:513:28)
    at emitErrorNT (node:internal/streams/destroy:157:8)
    at emitErrorCloseNT (node:internal/streams/destroy:122:3)
    at processTicksAndRejections (node:internal/process/task_queues:83:21)
Emitted 'error' event on DuplexWrapper instance at:
    at EventEmitter.emit (node:events:513:28)
    at ClientRequest.<anonymous> (D:\zry\dev\nodejs\betaflight-configurator\node_modules\got\index.js:73:7)
    at Object.onceWrapper (node:events:628:26)
    [... lines matching original stack trace ...]
    at processTicksAndRejections (node:internal/process/task_queues:83:21) {
  code: 'ETIMEDOUT',
  host: 'clients2.googleusercontent.com',
  hostname: 'clients2.googleusercontent.com',
  method: 'GET',
```

haslinghuis, 2023-02-02:
try yarn

smart2pet, 2023-02-02:
It's yarn.

haslinghuis, 2023-02-02:
ETIMEDOUT is a timeout. This happens also in Europe (sometimes).

Did you check:  https://betaflight.com/docs/development/Building%20in%20Ubuntu#building-betaflight-configurator

You could try changing DNS setting to use an open DNS server.

smart2pet, 2023-02-03:
So please make a solution that is able to use at all regions.

haslinghuis, 2023-02-03:
Maybe Onion Router or VPN ?

This is not a Betaflight problem as I would like to help any one I'm sorry not being able to help as this is outside our control.

closed 2023-02-03 by haslinghuis

smart2pet, 2023-02-03:
No. That's illegal in China.