betaflight/betaflight-configurator #3305: OSD view is too large in 10.9

dedrones, 2023-02-02:
### Describe the bug

Hi! I've updated my betaflight configurator to th 10.9 version and now the OSD tab shows the OSD preview to large
I'm scrolling up and down all time. Is this correct? Thanks!

### To Reproduce

Install Betaflight Configurator 10.9 and go to the OSD tab

### Expected behavior

OSD preview too large and not fixed to top

### Configurator version

10.9

McGiverGim, 2023-02-02:
A screen capture will help.

haslinghuis, 2023-02-02:
HD receives canvas size dynamically breaking the viewport at 1455px ???

dedrones, 2023-02-03:
> A screen capture will help.

A video could be better 
[Betaflight Big OSD Preview](https://vimeo.com/795508173/6c24468f50)

kevinsumner, 2023-02-13:
I'm able to reproduce this on macOS just by resizing the window horizontally. The first two screenshots I'm posting are taken seconds apart using macOS built-in window screenshot (⌘⇧5 then Space), with a slight expansion of the Configurator window in between. Video Format (including HD) doesn't appear to change this behavior at all. I'm using the virtual FC function to capture the screenshots since I don't have a real FC handy.

On the 13" M1 MacBook, I had to move to the highest resolution "More Space" option in the Displays settings panel. "Default" and anything else made it so that Configurator could never put the preview between the two settings columns, even while Configurator is full screened. According to SwitchResX, the "More Space" resolution is 1680x1050, while "Default" is 1440x900. *This is likely a workaround for anyone on a 13" MacBook with Retina Display.*

This behavior appears to be a regression in 10.9.0 -- 10.8.0 behaves as expected. 10.8.0 goes all the way to the minimum window size without reflowing the columns below the Preview panel.

## 10.9.0 with OSD settings columns flowed below Preview
[image]

## 10.9.0 with the normal flow
[image]

## macOS display panel settings for 13" M1 MacBook Pro.
[image]

## 10.8.0 at it's narrowest
[image]

kevinsumner, 2023-02-13:
I reproduced this on the 10.9.0 tag and then tried on master branch (currently at b7d1ec0837a42cede6b6e7d00a53ae27ff2995af). This appears to be fixed on master branch already.

closed 2023-03-24 by github-actions[bot]