betaflight/betaflight-configurator #3314: Link on VTX tab is broken (still points to GitHub wiki, which doesn't exist any more)

MrTucks, 2023-02-04:
### Describe the bug

The link to the vtx tables page on the VTX tab still points to the old wiki which doesn't exist anymore.

### To Reproduce

You don't need an FC
Connect using virtual mode (or connect to an FC that uses Smartaudio, Tramp)
Go to the Video Transmitter Tab
Click the link in the line "1. Go to this page;"
The link goes to https://github.com/betaflight/betaflight/wiki/VTX-tables but because the wiki doesn't exist anymore it redirects to https://github.com/betaflight/betaflight/ which is not helpful.

### Expected behavior

Link points to the new wiki. Maybe http://betaflight.com/docs/wiki/archive/VTX-Tables but I'm not sure. 

### Configurator version

10.9.0 (697943b)

### Flight controller configuration

```txt
Doesn't matter. You just need to be able to enter the VTX tab.
```

MrTucks, 2023-02-04:
The "Presets Wiki" yellow button in the top right of the Presets tab is broken in the same way as well.
Sorry if this was already reported a billion times.

haslinghuis, 2023-02-04:
After this one all links should be fixed: https://github.com/betaflight/betaflight-configurator/pull/3312

closed 2023-03-15 by github-actions[bot]