betaflight/betaflight-configurator #3315: Better cloud build options

MrD-RC, 2023-02-05:
### Is your feature request related to a problem? Please describe

Sorry, I don't like the cloud build at all. It will be especially confusing to new users. I also don't like that you will need to flash new firmware if you change some little hardware thing, like a receiver for example. I under stand it's there for F411 and F722 MCUs. But surely compiling some areas of the code for efficiency would have solved that problem, and not needed cloud build at all. 

Anyway. I have a couple of suggestions which may make the process better.

### Describe the solution you'd like

1. Have an **Install All** option next to the **Core Only** option. This will just install the full firmware in the old way. Much simpler if space is not an issue.
2. Allow selection of multiple radio protocols. For example if I want to install CRSF and FPORT, this is not possible now
3. Have separate telemetry options (again, multi-selectable) for all systems. CRSF and GHST not being in the list adds confusion. Also, I can't then install CRSF and SmartPort.

### Describe alternatives you've considered

Pretty self explanatory in the last section

MrD-RC, 2023-03-08:
No one else thinks that this can be improved?

haslinghuis, 2023-03-08:
You can add whatever you like, but radio, telemetry and motor protocol should be a single option.

However for your use case you can add to custom defines from this list:
```
SERIALRX_CRSF
SERIALRX_IBUS
SERIALRX_JETIEXBUS
SERIALRX_SBUS
SERIALRX_SPEKTRUM
SERIALRX_JETIEXBUS
SERIALRX_SUMD
SERIALRX_SUMH
SERIALRX_XBUS
SERIALRX_FPORT

TELEMETRY_CRSF
TELEMETRY_GHST
TELEMETRY_FRSKY_HUB
TELEMETRY_HOTT
TELEMETRY_IBUS
TELEMETRY_IBUS_EXTENDED
TELEMETRY_JETIEXBUS
TELEMETRY_LTM
TELEMETRY_MAVLINK
TELEMETRY_SMARTPORT
TELEMETRY_SRXL
```

MrD-RC, 2023-03-09:
Thanks for the info @haslinghuis But, I think it should be easier to use than having to edit custom defines. One note though. You said that "telemetry and motor protocol should be a single option.". But, the default telemetry is not a single option. It installs CRSF and GHST telemetry with the default settings. I don't really see why it needs to be singular either. Otherwise, if you change receiver, you could end up having to re-flash again, where as in 4.3 it would have just been changing an option.

I also think the **Install all** option would be useful. If I don't use an F411 or F722, I really can't be bothered with all this just to flash firmware.

haslinghuis, 2023-03-09:
Do you have more then once receiver installed on the flight controller and why? This is really an edge case. You still need to change settings etc.

Yes the CRSF, GHST, SBUS option should already have been removed.

aQuaplug, 2023-04-10:
I do hope option 'Install all'.
I have several radios and plenty of receivers. I like play and test with them. That's my hobby.
I don't want flash everytime when I swap some littlething.

And newbies in community of South Korea(my country :)) sometimes asks 'how to change receivers of second hand drone.'
, 'flashed for receiver, and motor spins but acts like crazy, what should I do T.T' (he don't knows about backup)
In BF 4.4. Newbies should know about their protocols(Yes, many newbies don't think about it), configuration backup and restore for just switch little thing. In many case, reply is 'Just using 4.3 or do not flash 4.4 if yours flying well'

I think classic way full build is still good and straight way, even cloud build is necessary for future.

sugaarK, 2023-05-24:
Point it with some chipsets you can’t build all because stuff already doesn’t fit.  The bf system has on average 6000 builds a day going through the system so it seems people are adapting and using the system just fine

MrD-RC, 2023-05-24:
> Point it with some chipsets you can’t build all because stuff already doesn’t fit. The bf system has on average 6000 builds a day going through the system so it seems people are adapting and using the system just fine

It doesn’t fit because everything is just optimised for speed. There is loads of code that could be optimised for efficiency. That would have not needed any of the cloud stuff. INAV is no doubt a bigger firmware. But that can still fit on F411 due to how it’s compiled.

haslinghuis, 2023-05-24:
> It doesn’t fit because everything is just optimised for speed. There is loads of code that could be optimised for efficiency. That would have not needed any of the cloud stuff. INAV is no doubt a bigger firmware. But that can still fit on F411 due to how it’s compiled.

Guess it's a trade off. Betaflight's main goal is flight performance.

Having categories (race, freestyle, cinematic, long range) we could choose the optimization level for specific needs.

The cloud API already allows addition of multiple radio and telemetry protocols using custom defines as I think this is still an edge case.

aQuaplug, 2023-05-25:
my case, adding multiple protocols with custom defines not working.
I added like this : SERIALRX_FPORT SERIALRX_IBUS SERIALRX_SPEKTRUM.
none of them working. only radio protocol works that selected in drop down menu.
And even if custom defines works, I'm still think this is not familiar to newbies.
It's common newbies learning FPV drone with second hand drone and replace receiver and fly.
4.4 firmware update UI is really not considered about newbies.

haslinghuis, 2023-05-30:
@blckmn After all the hard work to downsize the firmware should we consider a full build option for firmware size > 512KB ?

blckmn, 2023-05-30:
> @blckmn After all the hard work to downsize the firmware should we consider a full build option for firmware size > 512KB ?

This is available already for download from the releases itself. Anything with >512kb of flash space has everything compiled in. You need a complete dump to make the target operational. You don't need cloud build. It is merely a convenience.

Also, the original author has missed one of the big features of cloud building, which is the defaults for a target without reserving flash space for and needing to "download and set" defaults.

This becomes apparent in 4.5.0.

Prajwal-Koirala, 2023-12-27:
This is an issue about improving the Betaflight cloud build options. The author dislikes the current cloud build system because it requires flashing new firmware if a user changes some hardware, like a receiver. They also find it confusing for new users. The author suggests several improvements, including an “Install All” option that would install the full firmware without needing the cloud. They also suggest allowing users to select multiple radio and telemetry protocols. Other users chime in with their own suggestions and experiences. The discussion is ongoing.

I hope this is helpful!

nerdCopter, 2024-03-12:
+1 for full-feature option (by any name -- full, baseline, ...). i mentioned in the past as well, specifically with vendors in mind. but have now become accustomed to cloud-building. 
`make all` currently has no errors, so no reason not to.

hydra, 2024-03-12:
> +1 for full-feature option (by any name -- full, baseline, ...).

naming wise, as a name, 'factory', 'default' or 'baseline' works for me, 'full' does not as it implies everything is included when some code is not included unless specific defines are used.  I'm also not 100% convinced by the 'core' name, as it doesn't convey it's meaning or intent well since it's not really meant for end-users to use, same goes for 'cloud', perhaps 'custom' would have been better there IMHO, but that's a different story.

hydra, 2024-03-12:
Going back to the issue of users being confused by cloud builds, one major issue is that if they have a 'baseline'/'factory' firmware installed and try and add some 'non-baseline' feature, what happens is that they don't actually /add/ something, what happens is that they /remove/ loads of features and only add the additional feature they chose, so then all their other stuff that was working stops working until they figure out the exact features to add to the cloud build extra args input, which most users, in my experience, are not capable of doing, which leads to frustration and support requests.

haslinghuis, 2025-09-24:
- Updated cloud build API in https://github.com/betaflight/betaflight.com/pull/566