betaflight/betaflight-configurator #3316: Unified targets / Custom defaults are broken for EXST targets.

hydra, 2023-02-05:
### Describe the bug

For EXST targets, the firmware is not located in FLASH, but instead either RAM (H750) or external flash (H730).

The `ConfigInserter` is hard-coded to look at a specifc address:

ConfigInserter.js
```
const CUSTOM_DEFAULTS_POINTER_ADDRESS = 0x08002800;
```

This is incorrect.

Here is the output from the `.map` file generated for an H730 built with an external flash vma address of `0x901000000`.

```
.custom_defaults
                0x901fdfc0        0x8
                0x901fdfc0                . = ALIGN (0x4)
 *(.custom_defaults_start_address)
 .custom_defaults_start_address
                0x901fdfc0        0x4 C:\cygwin64\tmp\betaflight_STM32H730.elf.DFlUT1.ltrans2.ltrans.o
                0x901fdfc0                customDefaultsStart.lto_priv.0
                0x901fdfc4                . = ALIGN (0x4)
 *(.custom_defaults_end_address)
 .custom_defaults_end_address
                0x901fdfc4        0x4 C:\cygwin64\tmp\betaflight_STM32H730.elf.DFlUT1.ltrans2.ltrans.o
                0x901fdfc4                customDefaultsEnd.lto_priv.0
                0x901fdfc8                . = ALIGN (0x4)
                0x901fdfc8                __custom_defaults_internal_start = .
 *(.custom_defaults)
                0x901fdfc8                PROVIDE (__custom_defaults_start = __custom_defaults_internal_start)
                0x901fffc0                PROVIDE (__custom_defaults_end = (ORIGIN (CUSTOM_DEFAULTS) + LENGTH (CUSTOM_DEFAULTS)))
                0x90180f9f                _sidata = LOADADDR (.data)
```

If we look at the .bin file generated at offset 0x1fdfc0 ( 0x901fdfc0 - 0x901000000, __custom_defaults_internal_start - flash vma address)  we see this:

[image]

And we can see the bytes at at address correspond with:

```
0x901fdfc8                PROVIDE (__custom_defaults_start
```
and
```
0x901fffc0                PROVIDE (__custom_defaults_end = (ORIGIN (CUSTOM_DEFAULTS) + LENGTH (CUSTOM_DEFAULTS)))
```

### To Reproduce

attempt to flash any EXST target.

### Expected behavior

Location to insert custom defaults is found correctly and custom defaults are inserted correctly.

### Configurator version

`master` - `f3997bda`

### Flight controller configuration

```txt
N/A - applies to all EXST targets.
```

### Add any other context about the problem that you think might be relevant here

The hard coded method of attempting to find the custom defaults is fundamentally flawed.

When unified targets were first created I raised the fact that there should not be a hard-coded address and that instead a specific marker should be placed in the firmware which the configurator can find by scanning the entire address range used for flashing.  This is still the best long-term solution that I can think of.

to keep it backwards compatible, the block would contain:
1) the 4 byte start address, 4 byte end address,
2) a marker, e.g. 4 bytes, `CSTM`
3) a 4-byte CRC so the marker block can be validated.

The above requires a minor change to the firmware make files to inject a CRC, and a minor change to the linker scripts to include the 4 byte marker.

In the mean time a short-list of known-addresses needs to be added to the configurator.

hydra, 2023-02-05:
For an H750 EXST target a map file section is as follows:

```
.custom_defaults
                0x2407dfc0        0x8
                0x2407dfc0                . = ALIGN (0x4)
 *(.custom_defaults_start_address)
 .custom_defaults_start_address
                0x2407dfc0        0x4 C:\cygwin64\tmp\betaflight_STM32H750.elf.gTBS2f.ltrans2.ltrans.o
                0x2407dfc4                . = ALIGN (0x4)
 *(.custom_defaults_end_address)
 .custom_defaults_end_address
                0x2407dfc4        0x4 C:\cygwin64\tmp\betaflight_STM32H750.elf.gTBS2f.ltrans2.ltrans.o
                0x2407dfc8                . = ALIGN (0x4)
                0x2407dfc8                __custom_defaults_internal_start = .
 *(.custom_defaults)
                0x2407dfc8                PROVIDE (__custom_defaults_start = __custom_defaults_internal_start)
                0x2407ffc0                PROVIDE (__custom_defaults_end = (ORIGIN (CUSTOM_DEFAULTS) + LENGTH (CUSTOM_DEFAULTS)))
                0x2406f5e9                _sidata = LOADADDR (.data)
```

hydra, 2023-02-05:
I am prepping and testing a PR for a short-term workaround to look at a list of known-addresses.

hydra, 2023-02-06:
The situation is worse than I thought, there are two more bugs, as follows:

1) The `generateData` function in `ConfigInserter` does not insert the custom defaults into the memory region if that memory region is contained in the hex file, which it always will be for EXST targets as they are required to be the full length of the flash region being written AND contain a MD5 hash for verification which is computed on the entire firmware contents.  The configurator errors out with "Configuration area in firmware not free." in the development console, instead of overwriting that region with the config.
2) it does not re-calculate and update the MD5 hash required for EXST targets - if the custom defaults were injected the previously computed hash, computed by the firmware makefile, would be invalidated by the injection and the firmware will not be loaded by any bootloader as a result.

hydra, 2023-02-06:
For issue 3, above, Here is a screenshot showing the firmware built using `./support/scripts/build_spracingh7rf.sh`.

[image]

Here you can see the custom default range is: `0x901fdfc8` - `0x901fffc0`.
The hex contains data for address: 0x90100000 - `0x901FFFFF`,

The configurator code needs to be updated to inject the config, overwriting data if it exists, not just to add a new block like it does now.

For issue 4, above:

The H730 linker script defines this:
```
    OCTOSPI1_CODE (rx): ORIGIN = ORIGIN(OCTOSPI1) + 1M, LENGTH = 1M - _custom_defaults_size - _exst_hash_size /* hard coded start address, as required by SPRACINGH7 boot loader, don't change! */
    CUSTOM_DEFAULTS (r) : ORIGIN = ORIGIN(OCTOSPI1_CODE) + LENGTH(OCTOSPI1_CODE), LENGTH = _custom_defaults_size
    EXST_HASH (rx)    : ORIGIN = ORIGIN(OCTOSPI1_CODE) + LENGTH(CUSTOM_DEFAULTS) + LENGTH(OCTOSPI1_CODE), LENGTH = _exst_hash_size
```

and when you compile the H7RF firmware using the `obj\main\betaflight_STM32H730.map` file contains this:

```
.exst_hash      0x901fffc0       0x40
                0x901fffc0                . = ORIGIN (EXST_HASH)
                0x901fffc0        0x1 BYTE 0x0
                0x901fffc1        0x1 BYTE 0x0
                0x901fffc2        0x1 BYTE 0x0
                0x901fffc3        0x1 BYTE 0x0
 FILL mask 0x00000000
                0x901ffff0                . = ((ORIGIN (EXST_HASH) + LENGTH (EXST_HASH)) - 0x10)
 *fill*         0x901fffc4       0x2c 00000000
                0x901ffff0                __md5_hash_address__ = .
                0x901ffff0        0x4 LONG 0x0
                0x901ffff4        0x4 LONG 0x0
                0x901ffff8        0x4 LONG 0x0
                0x901ffffc        0x4 LONG 0x0
                0x90200000                . = (ORIGIN (EXST_HASH) + LENGTH (EXST_HASH))
                0x90200000                __firmware_end__ = .
```

And you can see the 16 byte MD5 hash need to be stored at address `0x901ffff0` as indicated by the `__md5_hash_address__` and which is set to zero by default - the `makefile` will sum it and inject it.

After the configurator is able to inject the custom defaults section it needs to re-calculate the MD5 hash which is done by generating an MD5 hash for the entire firmware image and a zero-filled EXST block.  i.e. the hash calculation needs to be given add all the bytes from the firmware except for the last 64 bytes, then given 64 bytes of zeros.  The total amount of bytes given to the hash algorithm is the same as the entire firmware size.

once the hash is calculated it then needs to be injected into the firmware structure/image/bytes at the offset indicated by the .map files and EXST specification here:  http://betaflight.com/docs/development/EXST-Bootloader

hydra, 2023-02-06:
There is another issue, we'll call it issue 5, as follows;

5) The configurator has no way of knowing if the .hex file being processed is an EXST target or not. And does not know if EXST hash recalculation needs to be performed.

Two potential solutions spring to mind:
a) distribute BF differently, supply a plain text .json file instead of a bin/hex file which describes the content and contains one or more base64 encoded gzip compressed 'binary' blob and the corresponding start address for each one, in this json also indicate that the binary blob is an EXST blob which the configurator can use to decide whether or not to update the hash.
b) further hack the workaround in #3321 of the list of known custom defaults address pointers to also include a flag saying that the address is for an EXST target, and use this flag to decide whether or not to update the hash.

a) should be considered for BF 5.x, b) will serve as a temporary workaround.

hydra, 2023-02-06:
It appears the long-term plan is to eradicate custom defaults entirely.

The plan is as follows:
4.4.1
* replace all the CLI commands with `#define`s in all the EXST unified targets.
* remove custom default support for EXST targets in the firmware.
* ensure all other CLI commands used in EXST targets actually have equivalent  `#define`s.  Currently motor outputs do not.

4.5.0
* do the same as above but for ALL targets
* remove all custom defaults code from the firmware.
* remove custom defautls support from configurator.
* update documentation to reflect the changes. (gui, targets repo, wiki, etc).

Confirmed by @blckmn on 2023/02/07 ~ 00:005 in the Discord server in the `#configurator` channel where this was discussed.

haslinghuis, 2023-06-21:
@hydra can you give an update regarding 4.4.2 and 4.5.0

hydra, 2023-06-21:
> @hydra can you give an update regarding 4.4.2 and 4.5.0

Sure!  Very simply: Ask @blckmn.  I'm not actioning this further as:
1) none of the PR's I raised following the above issue were accepted.
2) I was booted out from the #development channel on Discord server for asking questions that no-one else dared ask and thus had no further conversations regarding this since.
3) I have no desire or incentive.

blckmn, 2023-06-21:
Unified targets are no longer utilised in 4.5 so this issue can now be closed.

closed 2023-06-21 by blckmn

ps-igel, 2023-11-23:
@hydra @blckmn I came across this issue while trying to flash a local hex file for a BetaFPV 1S AIO provided through their website ([see here](https://support.betafpv.com/hc/en-us/article_attachments/12483028977945)). 
Configurator won't let me flash the file and only in debug can you see the reason being the "Configuration area in firmware not free." error. 

Whats the proper way to proceed here? I really need to flash this firmware.

haslinghuis, 2023-11-23:
If this hex is 4.2 or 4.3 could use portable version of configurator 10.8 to flash it.

ps-igel, 2023-11-23:
> If this hex is 4.2 or 4.3 could use portable version of configurator 10.8 to flash it.

10.8 doesn't work either unfortunately, neither does 10.7.1. Not sure what to make of this. There clearly must be a way to flash an old firmware with an old configurator right?