betaflight/betaflight-configurator #3331: GPS wont turn on with the base version firmware on the HKF7226 AIO

Molecuman, 2023-02-07:
### Describe the bug

I have a HKF7226 AIO with the base version and the latest Betaflight firmware/configurator and the GPS cannot be activated. It used to work before I installed the new firmware/configurator WITH Frsky Fport telemetry also working

### To Reproduce

Install the latest firmware/configurator on the HKF7226 AIO and connect a GPS on a spare UART with a Frsky Fport capable receiver in another UART port as Oscar Liang described here: https://oscarliang.com/setup-frsky-fport/

### Expected behavior

The GPS switch on the configuration panel remains ON and the GPS working and also the Frsky telemetry

### Configurator version

10.9.0

### Flight controller configuration

```txt
# 

# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.4.0 Jan 29 2023 / 03:02:23 (4605309d8) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name HAKRCF722D
manufacturer_id HARC

# feature
feature -TELEMETRY
feature -LED_STRIP
feature -AIRMODE
feature MOTOR_STOP

# serial
serial 0 8192 115200 57600 0 115200
serial 3 2 115200 57600 0 115200

# map
map TAER1234

# aux
aux 0 0 0 1300 2100 0 0
aux 1 1 1 900 1300 0 0
aux 2 2 1 1300 1700 0 0
aux 3 13 4 1700 2100 0 0
aux 4 28 0 1700 2100 0 0
aux 5 30 5 1700 2100 0 0
aux 6 39 3 1225 1450 0 0

# adjrange
adjrange 0 0 2 900 2100 29 2 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 5
vtxtable powervalues 0 25 200 400 800
vtxtable powerlabels PIT 25 200 400 800

# master
set dyn_notch_count = 1
set dyn_notch_q = 500
set acc_calibration = -21,8,29,1
set mag_calibration = -733,-2117,-1161
set fpv_mix_degrees = 25
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set failsafe_procedure = GPS-RESCUE
set small_angle = 45
set pid_process_denom = 2
set osd_cap_alarm = 450
set osd_tim_1_pos = 33
set osd_tim_2_pos = 6177
set osd_remaining_time_estimate_pos = 65
set osd_vtx_channel_pos = 22850
set osd_ah_sbar_pos = 4302
set osd_ah_pos = 4174
set osd_mah_drawn_pos = 6497
set osd_gps_speed_pos = 4385
set osd_gps_lon_pos = 4176
set osd_gps_lat_pos = 4161
set osd_gps_sats_pos = 4152
set osd_home_dir_pos = 6339
set osd_home_dist_pos = 6454
set osd_flight_dist_pos = 4438
set osd_compass_bar_pos = 4394
set osd_altitude_pos = 6188
set osd_avg_cell_voltage_pos = 6507
set osd_pit_ang_pos = 4278
set osd_rol_ang_pos = 4310
set osd_efficiency_pos = 6517
set osd_profile = 2
set osd_canvas_width = 30
set osd_canvas_height = 13
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 5
set vtx_freq = 5917
set vcd_video_system = NTSC
set gyro_1_sensor_align = CW90
set gyro_1_align_yaw = 900
set gyro_2_align_yaw = 900

profile 0

rateprofile 0

# end the command batch
batch end

#
```

### Add any other context about the problem that you think might be relevant here

The GPS works with the non base version firmware but the Frsky FPort telemetry dont work at all, even after playing with the serialrx_halfduplex and the serialrx_inverted ON and OFF (ON-ON, ON-OFF, OFF-ON and OFF-OFF). I also try to select "none", Smartport and Frsky_hub on the telemetry section with no success. Only with the base version the telemetry works but not the GPS
[image]

haslinghuis, 2023-02-07:
> feature -TELEMETRY

Telemetry seems to be disabled. GPS support is not include in Core Only build.

Basically the issue seems to be FPORT telemetry is not working as expected.
Please try flashing latest development build and let's have a look at the support ID you can grab at the CLI tab.

McGiverGim, 2023-02-07:
I think fport telemetry was fixed here: https://github.com/betaflight/betaflight/pull/12314
But the theory is that adding smartport as telemetry must work.

closed 2023-03-18 by github-actions[bot]