betaflight/betaflight-configurator #3336: Ports never finish "Loading..." on Ubuntu 22.04 and 22.10

Schlaegel, 2023-02-10:
### Describe the bug

Video demo below. Ports never finish loading on Ubuntu, so it is impossible to connect to the flight controller. In debug builds, manually restarting the app gets the ports to show and everything works.

### To Reproduce

This is 100% reproducable on my system. I did two fresh installs of Ubuntu to demonstrate it. I have included a video showing 10.8 working and the behavior I get on 10.9.

### Expected behavior

The ports would recognize the connected flight controller.

### Configurator version

10.9

### Flight controller configuration

```txt
I tried three different flight controllers. All were different, but all got the same results.
```

### Add any other context about the problem that you think might be relevant here

Here is a link to the video demo:[https://youtu.be/4hq73MRrnuw](url)

Schlaegel, 2023-02-10:
It looks like you will need to manually open that link:
https://www.youtube.com/watch?v=4hq73MRrnuw

haslinghuis, 2023-02-10:
> apt list --installed libatomic1 brltty modemmanager

Remove modemmanager.

Schlaegel, 2023-02-10:
The part you quoted shows that it isn't installed. I asked for it to list, and it showed it wasn't there. That was in Ubuntu 22.10.

I've done a lot of work to do fresh installs, and document everything so there isn't any confusion.

https://www.youtube.com/watch?v=4hq73MRrnuw

In the video demo, in Ubuntu 22.04, ModemManager and Braille TTY are installed, but masked, so they aren't running at all. The demo shows Configurator completely working on 10.8. The video should address all worries of misconfiguration.

In the video demo, in Ubuntu 22.10. ModemManager and Braille TTY are completely removed. The demo again shows the same results: Configurator 10.8 working and 10.9 not working.

I'm willing to do any checks/tests you want.

haslinghuis, 2023-02-10:
[image]

haslinghuis, 2023-02-10:
On master:

[image]

haslinghuis, 2023-02-10:
See also: http://betaflight.com/docs/development/Building%20in%20Ubuntu#building-betaflight-configurator

Schlaegel, 2023-02-10:
Did you see anything in my video that I missed from that document? I made the video to address those concerns up front.

I understand it is working on your computer. I also hope from watching the video, with my six different configurations, with fresh installs of Ubuntu, that you understand that it isn't working on my computer.

I'll work with you however, and I've done my best to assure you it isn't user error.

haslinghuis, 2023-02-10:
Please share console (press F12) on the loading screen.

Also like to see what happens after reload (again F12).

Schlaegel, 2023-02-10:
Thanks for working with this. I tried to use F12 on 10.9 and 10.10. I'm guessing it only works on debug builds, because it only did anything on 10.10.

Here is a log of it not working on 10.10:
[10.10-NotWorking.log](https://github.com/betaflight/betaflight-configurator/files/10711145/10.10-NotWorking.log)

Schlaegel, 2023-02-10:
I didn't close the Configurator, I just right clicked and Reload App:
[10.10-Reload-NowWorking.log](https://github.com/betaflight/betaflight-configurator/files/10711186/10.10-Reload-NowWorking.log)

haslinghuis, 2023-02-10:
Ok, this part is interesting as I don't see that all. Do you have wifi enabled flight controller or receiver? Are you using wifi?

Can you try with ethernet cable?

```
mdns_discovery.js:55 Found HTTP service Object
mdns_discovery.js:55 Found HTTP service Object
mdns_discovery.js:55 Found HTTP service Object
mdns_discovery.js:55 Found HTTP service Object
Error handling response: TypeError: Cannot read properties of undefined (reading 'vendor')
    at <URL>
    at Array.filter (<anonymous>)
    at <URL>
    at <URL>
```

Schlaegel, 2023-02-10:
My system is connected via ethernet directly to my router. I do have wifi on my system, but it isn't connected to anything.
ip a
`1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000`
`2: enp7s0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000`
`3: wlp5s0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN group default qlen 1000`

It might be worth noting I have a number of devices using mdns. I have many Google homes and printers, all using mdns.
avahi-browse -a
`+ enp7s0 IPv6 Project Room TV                               _androidtvremote2._tcp local`
`+ enp7s0 IPv4 Project Room TV                               _androidtvremote2._tcp local`
`+ enp7s0 IPv6 HP Color LaserJet MFP M283cdw (BD17F8)        Web Site             local`
`+ enp7s0 IPv4 HP Color LaserJet MFP M283cdw (BD17F8)        Web Site             local`
`+ enp7s0 IPv4 OKI-B430-A21C36                               Web Site             local`
`+ enp7s0 IPv4 elrs_rx_4CEBD60C9726                          Web Site             local`
`+ enp7s0 IPv6 _workstation                                  _tcp                 local`
`+ enp7s0 IPv6 _androidtvremote2                             _tcp                 local`
`+ enp7s0 IPv4 _workstation                                  _tcp                 local`
`+ enp7s0 IPv4 _services                                     _dns-sd._udp         local`
`+ enp7s0 IPv4 _http                                         _tcp                 local`
`+ enp7s0 IPv4 _androidtvremote2                             _tcp                 local`
`+ enp7s0 IPv6 _googlecast                                   _tcp                 local`
`+ enp7s0 IPv6 _googlezone                                   _tcp                 local`
`+ enp7s0 IPv4 _googlecast                                   _tcp                 local`
`+ enp7s0 IPv4 _googlezone                                   _tcp                 local`
`+ enp7s0 IPv6 homeassistant [e2] Workstation          local`
`+ enp7s0 IPv4 homeassistant [e2] Workstation          local`
`+ enp7s0 IPv6 Google-Home-Mini-4e _googlecast._tcp     local`
`+ enp7s0 IPv6 Google-Home-Mini-5a _googlecast._tcp     local`
`+ enp7s0 IPv6 Google-Home-Mini-b8 _googlecast._tcp     local`
`+ enp7s0 IPv6 Google-Home-Mini-ad _googlecast._tcp     local`
`+ enp7s0 IPv6 Google-Home-Mini-2a _googlecast._tcp     local`
`+ enp7s0 IPv6 Google-Home-2b  _googlecast._tcp     local`
`+ enp7s0 IPv6 google-cast-group-8a _googlecast._tcp     local`
`+ enp7s0 IPv6 google-nest-hub-f7 _googlecast._tcp     local`
`+ enp7s0 IPv4 Google-Home-Mini-ad _googlecast._tcp     local`
`+ enp7s0 IPv4 Google-Home-Mini-2a _googlecast._tcp     local`
`+ enp7s0 IPv4 Google-Cast-Group-b2 _googlecast._tcp     local`
`+ enp7s0 IPv4 Google-Cast-Group-a1 _googlecast._tcp     local`
`+ enp7s0 IPv4 Chromecast-5a   _googlecast._tcp     local`
`+ enp7s0 IPv4 D32f-G1-d8      _googlecast._tcp     local`
`+ enp7s0 IPv4 Chromecast-be   _googlecast._tcp     local`
`+ enp7s0 IPv4 Google-Home-2b  _googlecast._tcp     local`
`+ enp7s0 IPv4 Google-Home-Mini-4e _googlecast._tcp     local`
`+ enp7s0 IPv4 google-cast-group-8a _googlecast._tcp     local`
`+ enp7s0 IPv4 google-nest-hub-f7 _googlecast._tcp     local`
`+ enp7s0 IPv4 Google-Home-Mini-b8 _googlecast._tcp     local`
`+ enp7s0 IPv4 Google-Home-Mini-5a _googlecast._tcp     local`
`+ enp7s0 IPv6 4e          _googlezone._tcp     local`
`+ enp7s0 IPv6 5a          _googlezone._tcp     local`
`+ enp7s0 IPv6 b8          _googlezone._tcp     local`
`+ enp7s0 IPv6 ad          _googlezone._tcp     local``
`+ enp7s0 IPv6 2a          _googlezone._tcp     local`
`+ enp7s0 IPv6 2b          _googlezone._tcp     local`
`+ enp7s0 IPv6 f7          _googlezone._tcp     local`
`+ enp7s0 IPv4 5a          _googlezone._tcp     local`
`+ enp7s0 IPv4 ad          _googlezone._tcp     local`
`+ enp7s0 IPv4 2a          _googlezone._tcp     local`
`+ enp7s0 IPv4 be          _googlezone._tcp     local`
`+ enp7s0 IPv4 b8          _googlezone._tcp     local`
`+ enp7s0 IPv4 2b          _googlezone._tcp     local`
`+ enp7s0 IPv4 4e          _googlezone._tcp     local`
`+ enp7s0 IPv4 5a          _googlezone._tcp     local`
`+ enp7s0 IPv4 f7          _googlezone._tcp     local`
`+ enp7s0 IPv4 _http-alt                                     _tcp                 local`
`+ enp7s0 IPv4 _print-caps                                   _tcp                 local`
`+ enp7s0 IPv4 _ipp                                          _tcp                 local`
`+ enp7s0 IPv4 _ipps                                         _tcp                 local`
`+ enp7s0 IPv4 _pdl-datastream                               _tcp                 local`
`+ enp7s0 IPv4 _printer                                      _tcp                 local`
`+ enp7s0 IPv4 _scanner                                      _tcp                 local`
[... 19 lines cut ...]
`+ enp7s0 IPv4 googlerpc                                     _googlerpc._tcp      local`
`+ enp7s0 IPv4 HP Color LaserJet MFP M283cdw (BD17F8)        _http-alt._tcp       local`
`+ enp7s0 IPv4 HP Color LaserJet MFP M283cdw (BD17F8)        _print-caps._tcp     local`
`+ enp7s0 IPv4 HP Color LaserJet MFP M283cdw (BD17F8)        PDL Printer          local`
`+ enp7s0 IPv4 HP Color LaserJet MFP M283cdw (BD17F8)        UNIX Printer         local`
`+ enp7s0 IPv4 HP Color LaserJet MFP M283cdw (BD17F8)        _scanner._tcp        local`
[... cut ...]
`+ enp7s0 IPv4 _home-assistant                               _tcp                 local`
`+ enp7s0 IPv6 _home-assistant                               _tcp                 local`
`+ enp7s0 IPv4 Computer Monitor                              AirPlay Remote Video local`
`+ enp7s0 IPv4 Computer Monitor                              _viziocast._tcp      local`
`+ enp7s0 IPv4 Pi Home Assistant                             _home-assistant._tcp local`
`+ enp7s0 IPv6 Pi Home Assistant                             _home-assistant._tcp local`

haslinghuis, 2023-02-10:
@Schlaegel please test with fix above. This removes MSDN browser as default.

Curious if it now also works when enabled as the vendor field was not provided.

Schlaegel, 2023-02-11:
> Curious if it now also works when enabled as the vendor field was not provided.

Yes. It now works both ways. W00H000! Thanks!

I believe the abbreviation should be mDNS, and it needs adjusted in the Options UI.

Is it safe to use this, or should I wait for a point release to upgrade BetaFlight?

haslinghuis, 2023-02-11:
[image]

Yes have a few of them as I was a MCT.

haslinghuis, 2023-02-11:
> Is it safe to use this, or should I wait for a point release to upgrade BetaFlight?

Actually there already a few fixes in latest development version - as we try to fix issues as soon they pop up.

PR #3337 (2023-02-15): Fix mDNS browser
Fixes: #3336

Also mDSN browser in PortHandler is now disabled by default. Please use options tab to enable the feature.
closed 2023-02-15 by haslinghuis