betaflight/betaflight-configurator #3340: Build Configuration Profiles

kevinsumner, 2023-02-13:
### Is your feature request related to a problem? Please describe

As I'm flashing all my quads up to Betaflight 4.4, I find myself with one of a handful of build configurations that I reuse across a significant number of quads. They break down into two primary configurations, a bare-bones load out for F411-equipped tiny whoop AIOs, and dropping a couple options for freestyle quads. If I were flying long range, I'd add in Magnetometers on top of the default set.

### Describe the solution you'd like

I'd like to have the ability to save a selection of configuration as a profile for reuse across multiple flashes. This saves me from forgetting a specific option and helps ensure all my FCs are running the feature set I expect.

### Describe alternatives you've considered

For now, I'm continuing to manually-select the items I need in each flash and have reduced the configuration deviations I've made from the defaults to lower the chance of human error.

HThuren, 2023-07-17:
@haslinghuis I was wondering to fix this, but it seems to be done already ?

haslinghuis, 2023-07-17:
Fixed in https://github.com/betaflight/betaflight-configurator/pull/3332

If build key is present in firmware build options are preset.

closed 2023-07-17 by haslinghuis