betaflight/betaflight-configurator #3347: No OSD in DJI Goggles 2 w/ 4.4.0

Sean-Haverstock, 2023-02-19:
### Describe the bug

This may not be a bug, but I don't know the workaround if it isn't. With version 4.4.0 it seems there is no longer a `displayport_msp_serial` setting. I was not able to add it to my config. I could not get the OSD to display in my DJI Goggles 2. I reverted to 4.3.2 and was able to set `displayport_msp_serial` equal to 0. The OSD is now present in the display. 

### To Reproduce

Use DJI Goggles 2 with the latest firmware. 

### Expected behavior

The OSD to be on screen. 

### Configurator version

10.9.0

### Flight controller configuration

```txt
# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.3.2 Nov 28 2022 / 07:30:19 (60c9521) MSP API: 1.44

# config: YES

# start the command batch
batch start

board_name GEPRC_F722_AIO
manufacturer_id GEPR

# feature
feature -RX_PARALLEL_PWM

# serial
serial 0 1 115200 57600 0 115200
serial 3 0 115200 57600 0 115200

# beacon
beacon RX_LOST
beacon RX_SET

# aux
aux 0 0 1 900 1775 0 0
aux 1 1 0 900 1300 0 0
aux 2 2 0 1775 2100 0 0
aux 3 13 4 1700 2100 0 0
aux 4 35 3 1775 2100 0 0

# master
set dyn_notch_count = 1
set dyn_notch_q = 500
set acc_calibration = 44,-1,-39,1
set min_check = 1020
set sbus_baud_fast = ON
set blackbox_sample_rate = 1/16
set dshot_bidir = ON
set motor_output_reordering = 2,3,0,1,4,5,6,7
set align_board_roll = 180
set align_board_yaw = 0
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 3
set pid_process_denom = 2
set osd_tim_1_pos = 2200
set osd_flymode_pos = 2049
set osd_throttle_pos = 2433
set osd_crosshairs_pos = 2253
set osd_current_pos = 2167
set osd_mah_drawn_pos = 2136
set osd_craft_name_pos = 362
set osd_altitude_pos = 18817
set osd_warnings_pos = 14730
set osd_avg_cell_voltage_pos = 2072
set osd_battery_usage_pos = 34873
set osd_disarmed_pos = 2442
set osd_displayport_device = MSP
set debug_mode = GYRO_SCALED
set displayport_msp_serial = 0

profile 1

# profile 1
set motor_output_limit = 90
set feedforward_smooth_factor = 50
set dyn_idle_min_rpm = 35

rateprofile 1

# end the command batch
batch end
```

arrowcircle, 2023-02-25:
Same here. No OSD values are displayed. After disarm, I see stats headers, but without actual numbers

Sean-Haverstock, 2023-02-27:
Hi arrowcircle. I was able to get the OSD displayed. In the Ports tab enable Configurating/MSP on the UART that the Vtx is wired up to. Then in the peripherals tab of the same UART, select VTX (MSP + displayport). You can see more in this video. https://www.youtube.com/watch?v=Kti-6lWcJQU&t=453s

arrowcircle, 2023-02-27:
I made setup according to this video, but still nothing is shown in googles.

olikx, 2023-02-27:
Same exact problem right here, please let me know if you find the solution.

Sean-Haverstock, 2023-03-04:
I'm not sure which UART either of your Vtx's are wired up to, but the instructions in that video will only work if your Vtx is wired to UART1. Double check your wiring and select the correct UART. If you have the correct UART, I would check the OSD tab to see if you have actually selected any value to render. Maybe you don't have any values selected to display for the current profile?

closed 2023-03-04 by Sean-Haverstock