betaflight/betaflight-configurator #3354: betaflight-configurator_10.9.0_amd64.deb

kj7kpv, 2023-02-23:
### Describe the bug

Betaflight configurator 10.9.0 amd64.deb running on mint 20.02 linux seems to hang. Almost any changes in filters or PID changes cause 100% load average and seems to hang for several minutes to full hang.
I'm running on a Q9550 Intel CPU. Not having this problem with the 10.8 release. Any change of PIDS and the Disconnect never work and I usually have to "kill -9" to exit betaflight. Tried this on Ubuntu 18 as well as Mint 19. All exhibit this issue.

### To Reproduce

Do any PID slider change then goto filter make any change go back to the PID page then save. Try to go to configuration page. Should be hung.

### Expected behavior

Expect betaflight not to hang.

### Configurator version

latest release of betaflight for amd64.deb

### Flight controller configuration

```txt
Using all default settings. on a blitz F745 AIO, or RDQ Bardwell F7 flight controller.
```

Arne-W, 2023-04-02:
For me it behaves the same. When changing filters or PIDs the load increases to 90% on all cores. Sometimes the load drops back to normal and sometimes it stays at 90% and you have to kill the process.
I am on Kubuntu 22.04  and using the 10.9 portable release FC is a Mamba F722 App.

kj7kpv, 2023-04-02:
This behavior is in 10.8 also just not as bad. So I have moved all my quads and various flight controllers (iflight stacks, blitz, GepRC and others) back to 4.3 / 10.8 configurator amd64.deb. Wondering if this is an issue with the older hardware or just a general problem with the Linux versions? Any information on this would be appreciated. Wondering If I need to upgrade hardware but would rather not.

kj7kpv, 2023-04-02:
Hi Arne,
Thanks for your inputs. I would like to solve this problem. I'm experiencing this problem on all my flight controllers from iFlight F745 stacks, AIO's, GepRC F7XX AIO's, FOX Reaper AIO, Speedy Bee F7 Stack etc... Seems this is a configurator problem using Linux. I have tried various versions of Ubuntu and Mint from versions 18 to 22. All the same. I have gone back to configurator 10.8 and 4.3 firmware and that is much better although that hangs also once in a while. I'm wondering if its due to older hardware or just the linux OS. Be nice to get this figured out. What ever I can do to help.

Thanks again and best Regards,

Craig Jordan

Sent with [Proton Mail](https://proton.me/) secure email.

haslinghuis, 2023-04-02:
FYI: Since 4.3 slider (movements) are calculated using MSP calls over serial connection. Need good quality cables as I'm developing configurator for years now mainly using Ubuntu LTS every day.

kj7kpv, 2023-04-02:
Hi Mark,

I use several different cables as mini B ,C etc. Doesn't seem to make a difference. Curious I am using older hardware. CPU is a Q9550. Why it hangs some times and not others is perplexing. Even if I repeat exactly the same sequence. I will try to find a combo that I can repeat. Thank you for your information. I will exchange cables to see if it has effect.

Best Regards,
Craig Jordan

Sent with [Proton Mail](https://proton.me/) secure email.

Arne-W, 2023-04-03:
I did some further testing:
I used a new laptop with kubuntu 22.04 and a Intel coreI7 11th gen CPU. Everything else was the same - same USB cable and flight controller. I was **NOT** able to reproduce the behavior. Everything worked smooth and as expected.

Then I switched back to the system where I observed the issue. This system is an old Intel Core i5-4570 and was able to reproduce it again several times.
For me it does not depend on moving the sliders on one of the PID pages - I just have to stay a while on one of the PID pages and when I try to leave the PID page - switching to presets for example - all cores get high load and the configurator hangs. Saving the values works as expected. No other configurator page shows this issue.
Interestingly it does not happen if I don't stay on the PID page. If I open the PID page and just wait until all values are shown and then immediately switch to another page everything is fine. I have to stay a while on the PID page (10sek) to observe the issue and the longer I stay on that page the longer it hangs afterwards. 

I know that sounds strange somehow but everything I described was reproducible several times.
Let me know if there is any test or logging that could be done.

haslinghuis, 2023-04-03:
Can you please try the build here: https://github.com/betaflight/betaflight-configurator/pull/3397#issuecomment-1491024649

Setup a I5-4430 / 8GB / 128GB SSD with Ubuntu 22.04.2.LTS and could not reproduce

haslinghuis, 2023-04-03:
Keep looking :)

Issue seems to be in the THREE library

[image]

haslinghuis, 2023-04-03:
@Arne-W please test this build :point_up:

haslinghuis, 2023-04-03:
Tested on:

[image]

Arne-W, 2023-04-04:
@haslinghuis Yeehaaaaaa - works like a charm! It is much more responsive on the old Core i5 and the hanging issue when leaving the PID page is completely gone. Perfect!

Thanks for your effort!

kj7kpv, 2023-04-04:
Wow, thats great can't wait to try it.
-craig

Sent with [Proton Mail](https://proton.me/) secure email.

closed 2023-04-05 by haslinghuis