betaflight/betaflight-configurator #3365: Graphics glitches on Linux Mint 64 bit running xFce 64 bit

mitcht, 2023-02-28:
### Describe the bug

Upon loading Betaflight, the entirety of the program is riddled with crazy graphics glitches.   It affects the entire application, and makes it unusable.

I just updated from BF 10.8.0 so I could manipulate the new BF 4.4 I installed on my Roma F5 using a different computer.  10.8.0 was fine.

I use XFCE because it is ridiculously lightweight, and will run alright on my ancient Dell crap laptop in the field.

### To Reproduce

Install 10.9.0 onto Linux Mint + XFCE.  Look at all the crazyness.

### Expected behavior

Just regular functional Betaflight window interface.

### Configurator version

10.9.0

### Add any other context about the problem that you think might be relevant here

I'll post screenshots.

mitcht, 2023-03-01:
Switching my env using `startx -- :2 vt2 -extension GLX` to disable built in hardware acceleration allows the application to work properly.  I think there are issues with built in Linux hardware graphics acceleration.

haslinghuis, 2023-03-01:
Some system requires disabling gpu hardware acceleration as you found out.
https://github.com/betaflight/betaflight-configurator#graphics-issues

Not sure if xubuntu has the same glitch as have tested Ubuntu 22.04 and Fedora 35 (Gnome). Debian is also working.

closed 2023-03-01 by haslinghuis

cptX, 2023-08-15:
I can verify that the issue still remains in Linux Mint 21.2 (latest version) running Cinnamon.

haslinghuis, 2023-08-15:
@cptX did you check https://github.com/betaflight/betaflight-configurator-nightlies/releases

@nerdCopter is there a way to disable gpu acceleration (for configurator only in linux)?

Master configurator performs badly on 2th gen Intel with integrated graphics as well and found https://github.com/nwjs/nw.js/issues/8043 describing the issue.

mitcht, 2023-08-15:
> Switching my env using `startx -- :2 vt2 -extension GLX` to disable built in hardware acceleration allows the application to work properly. I think there are issues with built in Linux hardware graphics acceleration.
> 
> see https://unix.stackexchange.com/questions/408582/how-to-disable-hardware-acceleration-in-linux

mitcht, 2023-08-15:
@nerdCopter I'm pretty sure the hardware acceleration 'layer' in linux applies to the entire "layer" that renders window'ed applications.  I've not noticed a big difference after disabling it. YMMV.

nerdCopter, 2023-08-15:
package.json --> chromium-args , maybe `--disable-gpu`, but if it actually works is entirely different story.

`./package.json:7:` ` "chromium-args": "--disable-features=nw2 --disable-gpu",`

nerdCopter, 2023-08-15:
when i add `--disable-gpu`, then execute (`yarn gulp debug`), it still runs with animations and renders.  whether it is proper or not, i am unsure.

nerdCopter, 2023-08-15:
@mitcht , i run both Debian and Ubuntu exclusively with XFCE 4.X.  One with Intel integrated graphics, another with NVidia graphics.  I do not have same issues. I recommend looking into updated and/or proprietary graphics drivers.

nerdCopter, 2023-08-15:
@cptX , if you have acess, you can try a build of this here: https://github.com/nerdCopter/betaflight-configurator/actions/runs/5870934630
if artifacts do not show, i can upload somewhere.

cptX, 2023-08-16:
Hi guys, thanks for your quick reply. How is the betaflight configurator related to chromium? Is it a browser based implementation? In my computer it runs as an independent application, not in a browser...
I don't want to disable the graphics acceleration in my whole laptop/session. I want in the worst case to disable the acceleration only for this app.
Also, in Driver Manager of Linux there is no other driver suggested and as far as I know Intel gives opensource drivers to the linux community so the default driver should be the correct and best driver for this...

@nerdCopter I'm on vacations with limited internet speed. Download and building will be a difficult task but I understand there will be great hope if we have the gpu acceleration disabled

nerdCopter, 2023-08-16:
@cptX , Configurator uses NW.js framework and thusly essentially Chromium.  https://github.com/nwjs/nw.js/wiki

Additionally, i should note that maybe it is not desired to disable GPU-acceleration for everyone & all OS.

No rush, when you have time to test and report, please do let us know if it solved anything.

cptX, 2023-08-16:
Guys there is no need for recompilation in linux. Just find the executable and run it with the command line option --disable-gpu. This worked and all artifacts were gone but in my old laptop the cpu now often goes to 100%, but still the application is more useful than before, at least I can read and set parameters.

haslinghuis, 2023-08-16:
@cptX is that 10.9 or latest 10.10?

cptX, 2023-08-16:
tested with 10.9

haslinghuis, 2023-08-16:
Please also try 10.10