betaflight/betaflight-configurator #3366: Lack of Support for STM32F7X2 board on Betaflight Configurator 10.9.0

RyanMo20, 2023-03-03:
### Describe the bug

I am using Betaflight Configurator 10.8.0 because the new 10.9.0 Configurator doesn't have the STM32F7X2 board support that I need for my DIY flight controller. Betaflight Configurator 10.8.0 is working okay, but I want to be able to update my board to BF4.4 and I need Configurator 10.9.0 to support my board before I can do so.

### To Reproduce

When flashing firmware in Betaflight Configurator 10.9.0 there is no board option labeled STM32F7X2. I am stuck using Configurator 10.8.0 because that is the latest version that has the STM32F7X2 board support.

### Expected behavior

I did not realize Betaflight Configurator 10.9.0 would not include the STM32F7X2 board that had previously been included.

### Configurator version

10.9.0

haslinghuis, 2023-03-03:
STM32F7X2 is not a board but MCU target. A board uses an MCU target with a board configuration.

What brand and model is your board, link please.

RyanMo20, 2023-03-03:
It is a custom board I made, and I used the STM32F7X2 MCU Target and a custom config file for the firmware.

haslinghuis, 2023-03-03:
Look at src/config/board/config.h

    #define FC_TARGET_MCU

closed 2023-03-03 by haslinghuis

RyanMo20, 2023-03-04:
How do I flash my flight controller if I cannot select "STM32F7X2" under "board" in Configurator Firmware Flasher? Since the generic STM32F7X2 is no longer listed under "board" in Configurator 10.9.0 (which it was in Configurator 10.8.0) do I have to go back to Configurator 10.8.0 every time I need to re-flash my board? Any help is appreciated.

beqarion, 2023-03-29:
I have the same issue. I have diatone mamba f722 with 
# version
# Betaflight / STM32F7X2 (S7X2)
I tried firmwares with name including 'f722' before, but it never worked. I used good old 'STM32F7X2 '. So Im going to go back to bf 10.8.0 now and I hope it will work again.

haslinghuis, 2023-03-29:
@beqarion which diatone board do you have?

RyanMo20, 2023-03-29:
@haslinghuis What should I do about my custom board? Will the STM32F7X2 MCU Target be brought back onto the configurator firmware flasher page, or is there another solution? Thank you

haslinghuis, 2023-03-29:
@RyanMo20 you need to create a board config under src/config

beqarion, 2023-03-29:
> 

mine is mk1, with micro usb. more exactly, its ' Diatone Mamba f722s mk1'

RyanMo20, 2023-03-29:
@haslinghuis Alright sounds good, thank you for the response!

haslinghuis, 2023-03-29:
@beqarion not sure but MAMBAF722 could work

beqarion, 2023-03-29:
> @beqarion not sure but MAMBAF722 could work

I know exactly that I flashed with exactly that board name and it didnt work with my current config(this config was backed up on firmware with same version). Everything was working except  betaflight couldnt 'see' controller inputs, altough tx module was connected to the controller.

haslinghuis, 2023-03-29:
Did you try core build for testing?

beqarion, 2023-03-30:
2,3 months in the hobby, dont know that much yet.

christhomas, 2023-09-05:
I also have this problem too with the emax hawk apex using a board which has no target that I can pick from betaflight, although emax did eventually give me a hex file and two betaflight cli dumps to flash with. I can't see how I can put betaflight 4.4 onto this board since what board target do I use from the dropdown to apply to my board.

What I thought was that firmwares were created for various mcu types and you'd use betaflight cli dumps to configure the resources. So I could use a generic stm32f7x2 mcu and then configure everything using a cli dump. But I don't know if that can work or not.

nerdCopter, 2023-09-06:
> 2,3 months in the hobby, dont know that much yet.

[image]
maybe helpful: https://hackmd.io/@nerdCopter/ByheaoLQh

nerdCopter, 2023-09-06:
> [...] generic stm32f7x2 [...]

with Betaflight 4.4 and into the future, unified-targets has gone away.  However, you might still find those in the Github releases "Assets".

christhomas, 2023-09-06:
So betaflight is producing targets for known branded boards and providing a configuration per target? So no more generic targets that I can select and then configure the resources for customised boards?

nerdCopter, 2023-09-07:
> So betaflight is producing targets for known branded boards and providing a configuration per target? So no more generic targets that I can select and then configure the resources for customised boards?

please see: https://github.com/betaflight/config (you could create a custom config.h and compile your own)

please join [Betaflight Discord](https://discord.betaflight.com/) for further conversation.