betaflight/betaflight-configurator #3372: Disclose GPS override Instructions

BadpawFPV, 2023-03-07:
### Is your feature request related to a problem? Please describe

When using GPS with Betaflight, the only way to get the drone out of Return to Home mode when activated by a failsafe, is to push both sticks to over 30%. No one actually knows this. This is critical satiety information which should be disclosed on the GPS page for safety  reasons

### Describe the solution you'd like

On the configutator GPS page, provide RTH override information in text. Both sticks must be moved over 30% to override a Return to Home activated by a failsafe. 

### Describe alternatives you've considered

Eliminate the 30% stick threshhold entirely, and use switch to cancel Return to Home 

### Other information

I asked several experienced pilots and sponsored pilots, along with several drone builders about how to cancel GPS return to home. I would classify this as unknown safety information, despite the mention deep in the Oscar Liang article

KTM1551, 2023-04-12:
i think this should definetely be done, id much rather have a cancel button and be ale to use the sticks to add slight corrections to the drone as it lands.

thanks
anthony

Prajwal-Koirala, 2023-12-27:
**⚠️ Attention Betaflight Developers: Prioritize Safety with Clear RTH Override Instructions ⚠️**

**I strongly advocate for incorporating clear and visible RTH override instructions directly within the GPS page of Betaflight Configurator.** This crucial safety information is currently overlooked, potentially leading to hazardous situations.

**Here's why this change is imperative:**

- **Widespread Lack of Awareness:** As highlighted in user feedback, even experienced pilots are often unaware of the 30% stick threshold for RTH override.
- **Critical for Failsafe Scenarios:** Understanding this override mechanism is essential for regaining control in unexpected failsafe activations.
- **Safety First:** Ensuring pilots have this knowledge can help prevent accidents and promote responsible flight practices.

**Proposed Implementation:**

- **Concise and Highlighted Text:** Display a clear message on the GPS page, stating: "To override RTH activated by failsafe, move both sticks over 30%."
- **Visual Aid:** Complement the text with an informative image or diagram illustrating the stick movement required for override.
- **Contextual Placement:** Position this information prominently near relevant settings, such as RTH mode or failsafe configuration.

**Additional Considerations:**

- **Explore Alternative Override Methods:** Investigate the feasibility of switch-based RTH cancellation as a user-configurable option, providing greater flexibility.
- **Incorporate into Documentation:** Include detailed RTH override instructions in official Betaflight documentation and resources.

**By prioritizing this change, we can collectively enhance safety awareness and empower pilots to confidently navigate failsafe scenarios. Let's make Betaflight even more robust and user-friendly!**

****

KTM1551, 2024-04-13:
> **⚠️ Attention Betaflight Developers: Prioritize Safety with Clear RTH Override Instructions ⚠️**
>
> **I strongly advocate for incorporating clear and visible RTH override instructions directly within [...]

Ive been using the GPS Land feature in BF from the very beginning when i got the GPS on there. 
I had no idea you could take control back after an actual F.S. triggered auto-RTH... I guess i never moved both sticks the few times its happened, without me triggering it.

Also I see most people don't even have a manual switch assigned for "RTH". Which should maybe be required or highly recommended. That way if you lose video feed or have some controller stick issue or for whatever reason, you can switch the drone into the RTH Mode and let the Drone gyro and GPS take over.. 

I also have my Failsafe settings set so that when I'm flying normally, Acro/Air Mode, its set to change to Angle mode on a Fail Safe or RTH, and i have it set for Fixed Height return. 
This makes the settings setup a lot easier and a lot more familiar for people that aren't super familiar with BF yet. 

Also after making the changes and then doing the fine-tuning on the other settings, the actual flying of the Drone on RTH mode, watching it in the Goggles fly back to my Home spot on its own, it works very good and really does it just as good as my DJI drone especially with the M10 chip now. Obviously an FPV Drone cant see Obstacles or see the Ground for landing, so thats the biggest difference.. 

I did have to tweak the settings for mine.. 
First i lowered the Descent speed because it was too fast and made it jerky.
I lowered the H/S slightly as well, on Return, for the same reason
But the thing that helped the most was making the Descent Height down, 1-2ft more than my Actual Set RTH Altitude, that way the Drone wouldn't disarm itself half a foot or more off the ground, which is sometimes concrete. So now it lands on the ground gently and with the motor disarm time raised 0.5secs it shuts off at the perfect time when its completely done landing but no extra time, which creates dirt & dust getting in the drone..

haslinghuis, 2024-04-13:
See: https://betaflight.com/docs/wiki/guides/current/GPS-Rescue-v4-5

KTM1551, 2024-04-15:
> See: https://betaflight.com/docs/wiki/guides/current/GPS-Rescue-v4-5

Is betaflight v4.5 and official release yet or has it been it pretty well tested by this point? 
I just updated 2 of my older drones to BFv4.4.3 last week, and that was the newest FW that was under the Stable list. I saw BF V4.5 but it wasnt on that list yet..

haslinghuis, 2024-04-15:
Start here: https://github.com/betaflight/betaflight-configurator/releases/tag/10.10.0-RC4 and select release candidate in firmware flasher to be able to test latest firmware release candidate.

haslinghuis, 2024-05-05:
Just use > 30% stick deflection to override GPS Rescue. It even should land in 4.5 if configured correctly.

closed 2024-05-05 by haslinghuis