betaflight/betaflight-configurator #3373: windows 7 help for new betafl.config issues

FlyingBuzzard, 2023-03-07:
Install it and try if flight controller conects:
https://anonfiles.com/7fWdR136w0/betaflight-configurator-installer_10.8.0_win32_zip

_Originally posted by @Asizon in https://github.com/betaflight/betaflight-configurator/issues/2707#issuecomment-999888776_

tried this link and it downloaded BUT it states: failed to load extensions from c program files beta flight config localization used but default_ locale wasn't specified in the manifest 

WHAT does that mean ? and how can I fix it to get the configurator  on my Window 7 PC

HELP

ryet9, 2023-03-11:
try using the PORTABLE version.. just unzip and run in place.

Pancronos, 2023-03-14:
Windows 7 reached End Of Life (EOL) on 14 January 2020. My suggestion if you still want to use such hardware is convert it to linux.

closed 2023-03-14 by haslinghuis