betaflight/betaflight-configurator #3387: Inability to Flash Firmware to my STM32F7X2 Flight Controller on Betaflight Configurator 10.9.0

RyanMo20, 2023-03-19:
### Describe the bug

I cannot use Betaflight 10.9.0 to flash my flight controller because there is no longer the generic "STM32F7X2" MCU Target on the dropdown list under "Board" in the firmware flasher. I made a custom flight controller and I need the generic MCU Target to flash, along with the custom config file I made. I don't want to have to revert back to Betaflight 10.8.0 Configurator every time I have to flash my flight controller. Any help is appreciated.

### To Reproduce

When flashing firmware in Betaflight Configurator 10.9.0 there is no longer the MCU Target option labeled "STM32F7X2" in the dropdown list under "board" in the firmware flasher tab. I am stuck reverting to Configurator 10.8.0 whenever I need to flash because that is the latest version that has the STM32F7X2 MCU Target support.

### Expected behavior

I expected Betaflight Configurator 10.9.0 to include the STM32F7X2 MCU Target that had previously been included, and is what I use for my custom flight controller.

### Configurator version

10.9.0

beqarion, 2023-03-29:
I have the same issue. Now Im gonna install old version of bf and see if Im able to find STM32F7X2  version there.

haslinghuis, 2023-03-29:
Duplicate of #3366

closed 2023-03-29 by haslinghuis

RyanMo20, 2023-03-29:
@haslinghuis Hi Mark, I understand this is a duplicate. I made it again because I never got a solution or answer to my question before you closed the last one. I would greatly appreciate some sort of solution. I'm not sure if the STM32F7X2 MCU Target is going to be brought back, or if there is some other solution. Thank you

Ryan