betaflight/betaflight-configurator #3393: Selection of Port doesn't recognize Flight Controller Port

mark-purnell, 2023-03-24:
### Describe the bug

With 10.9.0 (697943b) Betaflight does not display the com port that it is reading from and the selection dropdown just reads "Loading....".    Happens on numerous flight controllers and doesn't happen with 10.8.0   

### To Reproduce

start 10.9.0
Connect Fight controller to computer
hear UBS tones and verified that STM appears COM3
Installed the lastest CP210x Drivers, Zadig and ran the Impulse RC Driver Fixer all success
Upper right corner device selector stuck at "Loading...."

### Expected behavior

Connect to Flight controller successfully 

### Configurator version

10.9.0 (697943b)

haslinghuis, 2023-03-24:
Please try latest nightly as configurator is probably waiting on mdsn devices on your network.
With Windows 10 there is no need to install drivers.
In some cases IRC Driver Fix or Zadig may fix some issues.

mark-purnell, 2023-03-24:
Not sure what you mean by trying the lastest nightly...  A newer build of BF?      What MDSN devices would it be waiting for?