betaflight/betaflight-configurator #3411: Ubuntu 22.04 : "Flash Firmware" not possible

fpvbam, 2023-04-08:
### Describe the bug

Firmware successfully loaded [Online] or cached.
Executing "Flash Firmware" prompts "Please load firmware file".

### To Reproduce

Select DFU, Update Firmware, Load Firmware [Online], Flash Firmware

### Expected behavior

Firmware flashing should be processed.

### Configurator version

10.9.0 (697943b)

haslinghuis, 2023-04-08:
Check http://betaflight.com/docs/development/Building-in-Ubuntu#flashing-a-target-with-betaflight-configurator-on-ubuntu-2204

fpvbam, 2023-04-08:
Thank you for the hint. This is a development machine, udev rules already added.

haslinghuis, 2023-04-08:
No problems here on master

haslinghuis, 2023-04-08:
Any output on dev console?

fpvbam, 2023-04-09:
Right now I cannot try further.
The F722 of the iflight TwinG F7 does not offer DFU anymore.
Neither ST-Link with soldered SWDIO/SWCLK wires will work.
I will replace the FC soon.

closed 2023-04-09 by fpvbam