betaflight/betaflight-configurator #3429: Custom logo image selection not working on MacOS

martinschaer, 2023-04-23:
### Describe the bug

When I choose an image to change the logo on the OSD, the preview image does not update.

The file system dialog works, but when I choose the image and it closes, the preview image is not update. My guess is that on MacOS, the `file://{path}` is not working.

Here is a minimal example of another way of doing this:

```html
<!DOCTYPE html>
<html>
  <head>
    <title>Minimal example to open and preview an image</title>
    <script type="module">
      const fileInput = document.querySelector("input");
      const imageEl = document.querySelector("img");
      const buttonEl = document.querySelector("button");

      // Button trigger the hidden native file input
      buttonEl.addEventListener("click", () => {
        fileInput.dispatchEvent(new MouseEvent("click"));
      });

      fileInput.addEventListener("change", (event) => {
        const file = event.target.files[0];
        const reader = new FileReader();
        reader.onloadend = () => {
          imageEl.src = reader.result;
        };
        if (file) {
          // readAsDataURL instead of: imageEl.src = `file://${path}`;
          reader.readAsDataURL(file);
        } else {
          imageEl.src = "";
        }
      });
    </script>
    <style>
      button {
        padding: 0 9px;
        background-color: #fb0;
        border: 1px solid #dba718;
        border-radius: 3px;
        font-weight: bold;
        text-shadow: 0 1px rgba(255, 255, 255, 0.25);
        line-height: 28px;
      }
      button:hover {
        background-color: #ffcc3e;
      }
    </style>
  </head>
  <body>
    <!-- Hidden file input, triggered by the button -->
    <input type="file" style="display: none" />

    <!-- Open file system modal by triggering the input above -->
    <!-- Common trick to add custom style, because file inputs are not fully customizable -->
    <button>Open file</button>

    <!-- Preview image -->
    <img src="" />
  </body>
</html>
```

Let me know if this is a good approach. I can help with a PR :) but I would need to set the env first.

### To Reproduce

On MacOS, choose a custom logo from the file system.

### Expected behavior

The preview image should update.

### Configurator version

10.9.0

martinschaer, 2023-04-28:
I found out the error was the color map on my image… I found out how to show the logs and there I saw the error.

closed 2023-04-28 by martinschaer

SitrucL, 2023-08-13:
Hi @martinschaer, how did you enable or display the debugging logs, I think I may be experiencing the same issue but not too sure.

martinschaer, 2023-08-18:
@SitrucL there is a button on the app to display the debugging logs. But as I mentioned, my error was on the image I created. It's hard to make an image with the correct color map