betaflight/betaflight-configurator #3432: Bookmarks for TCP connections

Yith1, 2023-04-27:
### Is your feature request related to a problem? Please describe

It would be great to be able to save bookmarks of the different tcp connections to my various ELRS 3.0 Wifi enabled quads.

It's great to be able to configure BF through ELRS over wifi, but having to type in the IP address of the ELRS RX all the time is annoying. Adding a bookmarking feature would be good.

### Describe the solution you'd like

I have a set of bookmarks to my different quads in my Chrome bookmarks bar which I use when flashing or configuring the ELRS RX itself. Something like that is what I'm after in BF Configurator.

### Describe alternatives you've considered

N/A

Yith1, 2023-04-27:
In addition if BF Configurator could actually ping the TCP connections it has saved then it could even detect when the RX was seen and filter the list... or even auto load up the config like it does with a USB connection!

Prajwal-Koirala, 2023-12-27:
** Bookmarking TCP Connections for Streamlined ELRS 3.0 Over WiFi Workflows **

**I enthusiastically endorse the addition of TCP connection bookmarking to Betaflight Configurator.** It would significantly enhance the user experience for those working with multiple ELRS 3.0 WiFi-enabled quads.

**Here's how this feature would streamline workflows:**

- **No more manual IP address entry:** Effortlessly connect to frequently used quads with a single click.
- **Quad management made simple:** Organize and access bookmarks effortlessly, saving time and frustration.
- **Enhanced efficiency and convenience:** Focus on configuration, not connection logistics.

**Envisioning a User-Friendly Implementation:**

- **Dedicated "Bookmarks" section** within the Configurator for easy management.
- **Ability to create, edit, and delete bookmarks** for flexibility and customization.
- **Storage of IP address, quad name, and optional notes** for clear identification.
- **Seamless connection initiation** directly from bookmarks.
- **Consider visual cues** (e.g., icons, colors) to differentiate bookmarked quads.

**Potential Additional Benefits:**

- **Integration with other features:** Imagine bookmarks triggering specific configuration profiles or OSD layouts for a truly personalized experience.
- **Cross-platform compatibility:** Ensure bookmarks sync across devices for seamless workflows.

**I'm confident that this feature would be warmly welcomed by the ELRS community. Let's make quick and convenient connections a reality!**

Witty-Wizard, 2025-02-04:
Question: With the new PWA Implementation does it support TCP?

haslinghuis, 2025-02-04:
Yes - using a web socket proxy (is needed for SITL) - but also looking at [webtransport](https://developer.mozilla.org/en-US/docs/Web/API/WebTransport)