betaflight/betaflight-configurator #3433: Motor direction wizard spins up all motors without asking

evilC, 2023-04-30:
### Describe the bug

I have seen this happen twice now...
Both times, I selected `Reorder Motors > Individually` and when the dialog popped up, instead of having `1` `2` `3` `4` buttons, there was only a `1` button.
So I `X` out of the dialog, and on doing so, ?all? motors spun up (I don't know how many, could have been all but motor 1)
Maybe it's something silly like, in some edge case where it is only going to display one button, the other unused buttons have not been laid out properly and are overlapping the `X`, so when you click it, you are actually clicking multiple motors too?

### To Reproduce

Cannot reliably repro

### Expected behavior

The quad does not try and eat my face

### Configurator version

10.9.0 (697943b)

closed 2023-06-07 by github-actions[bot]

raster74, 2024-07-06:
Still happening on Configurator 10.10.0.  You click on "Motor dIrection" and confirm that you understand the risks, select "Individually", and instead of having all of the motor buttons (eg. 1,2,3,4 for a quad), you're presented with one button that says "All".  If you don't click on the button, I've had cases where all motors have spun up and others where just Motor 1 spins up to a fairly high rpm.  

This has happened to me about 4-5 times now, twice this morning with 10.10.0 and previously with 10.9.0

haslinghuis, 2024-07-06:
[image]

[image]

WIZARD:
[image]

INDIVIDUALLY:
[image]

MadfishFPV, 2024-07-07:
Mark, not sure what you’re trying to convey here with the photos but the bug is a transient bug, which occurs every so often not always.   The last photo you presented shows options to spin motors 1 2 3 4 or All.   Occasionally you don’t get this and the only button is All and then suddenly one or all of your motors spin up.   As I’ve said I’ve had this numerous times now even with the latest configurator.   I’ll try updating the quad that this most recently happened with to latest BF firmware as it’s not on 4.5.

haslinghuis, 2024-07-07:
> Occasionally you don’t get this and the only button is All and then suddenly one or all of your motors spin up. 

@MadfishFPV motors start spinning without clicking any button?

@limonspb any idea about this intermittent behavior.

nerdCopter, 2024-07-08:
jsut wondering if this is BLHeli 32.9 specific or not?

MadfishFPV, 2024-07-08:
> > Occasionally you don’t get this and the only button is All and then suddenly one or all of your motors spin up.
> 
> @MadfishFPV motors start spinning without clicking any button?
> 
> @limonspb any idea about this intermittent behavior.

Yes doesn't start spinning straight away, usually have about 10 seconds before it happens so when I saw it happen on the weekend again I was in panic modes as I didn't have a smoke stopper attached and this usually trips as the motor or motors go full throttle.   While it's considered a no no, I leave props on with quads under 4 inches while checking motor direction, so was a fun experience 😂.

MadfishFPV, 2024-07-08:
> jsut wondering if this is BLHeli 32.9 specific or not?

Not Bheli32, don't think any of my stacks or aio are bheli32,  majority if not all that I have arrive as BheliS which I swap to BlueJay 0.19, don't think I've tried 0.21 or 0.22 yet on any of my quads as I know there was an issue with 0.20 and haven't been keen to try the latest yet

MadfishFPV, 2024-07-08:
> For BlueJay we still recommend using version `0.19.2`
> 
> https://betaflight.com/docs/wiki/getting-started/hardware/esc-firmware#bluejay

Yes I meant 0.19.2, the last one before 0.20 was yanked as it apparently toasted some esc / aio.   I run 0.19.2 on all of my quads

limonspb, 2024-07-09:
unless something changed in configurator, im not sending any commands unless user clicks a button. I'd love to have a video of a bug or something. Plus a full setup/config

evilC, 2024-07-09:
I did try and see if I could get it to reliably repro, but no joy.
Unfortunately, I think it's gonna be a heisenbug :(
I am solely on BL_S / BlueJay too, so cannot comment if it happens with BL32

limonspb, 2024-07-09:
> I did try and see if I could get it to reliably repro, but no joy. Unfortunately, I think it's gonna be a heisenbug :( I am solely on BL_S / BlueJay too, so cannot comment if it happens with BL32

for the record of this issue when this occurs again (if cant record, cuz obviously you can't record all the time) do the following:
1. Share here your diff, FC name, ESC name, ESC firmware (double check)
2. Share the exact steps you did like 1,2,3
Thanks!