betaflight/betaflight-configurator #3436: Firmware Flasher page 'USB Flashing' help link not working

junwoo091400, 2023-05-04:
### Describe the bug

Current 'USB Flashing' link in the configurator's Firmware flasher page points to https://github.com/betaflight/betaflight/wiki/Installing-Betaflight, which doesn't exist anymore.

It should rather point to: https://betaflight.com/docs/wiki/boards/BETAFLIGHTF3#driver-related-problems

### To Reproduce

[image]

In the configurator, click the 'USB Flashing' text above

### Expected behavior

Should direct users to the appropriate page where the driver installation steps are introduced: It should rather point to: https://betaflight.com/docs/wiki/boards/BETAFLIGHTF3#driver-related-problems

### Configurator version

10.9.0

### Add any other context about the problem that you think might be relevant here

I couldn't really find the central source of truth for where to download the appropriate drivers, so if there's a better link, that would be better!

junwoo091400, 2023-05-04:
Note, referencing https://www.reddit.com/r/Multicopter/comments/w9qo32/unable_to_flash_firmwareget_stuck_at_initiating/, this could be worth adding to the helpful programs for driver related fixes as well: https://impulserc.com/pages/downloads

haslinghuis, 2023-05-04:
The link is fixed in the nightly builds of configurator. Now each tab has it's own wiki page:
https://betaflight.com/docs/wiki/configurator/firmware-flasher-tab

https://betaflight.com/docs/wiki/archive/Installing-Betaflight

closed 2023-05-04 by haslinghuis