betaflight/betaflight-configurator #3440: Gray out features that were not complied during cloud complie

haybeseret, 2023-05-06:
### Is your feature request related to a problem? Please describe

Hello,
So the featuer is like so.
When a FC is flashed with cloud build, it could be nice to add an option / log which tell us which features were selected, and when this FC is connected to the configurator to gray out the disabled features.
I am writing this after spending 2 days figuring out why GPS is not tunred-on on a FC that was flashed without GPS .
Thanks.

### Describe the solution you'd like

Have UUID per flash / or similar, that will be stored either on the local flash or in the cloud which will hold which features were flash on the FC.
When FC is connected to the configurator, have that info determine which features whould be seen as enabled.

### Describe alternatives you've considered

Try to remember which features were flash ?!

haslinghuis, 2023-05-06:
Try: https://github.com/betaflight/betaflight-configurator-nightlies

McGiverGim, 2023-05-06:
I think that the idea of gray them (or any other color) is better than hide them. In this way the user knows that the feature exist but it's not available.

haybeseret, 2023-05-07:
Hi - tested lastest nightly  and indeed seem like it was fixed. yet the UX is not optimal. Need to still grayout with tooltip all the locations were the feature iss relevant - GPS for exmaple, need to also be disable in the sensors dropdown in the ports tab.
Thanks.

mheidinger, 2024-03-18:
One additional thing that would be nice to be included: Hide / grey out not compiled receiver protocols.

I wanted to move my RC stuff to ELRS and could not figure out why it was not working on my quad. Turns out I just did not include `USE_SERIALRX_CRSF` in my build and completely forgot that I selected something there a few weeks back. Took more time than I'd like to admit to figure this out (ultimately installed 10.10.0 and used the build key from the log to query the build server directly to get 'my' config).

haslinghuis, 2024-03-18:
- I think this PR https://github.com/betaflight/betaflight/pull/13333 could make it possible to implement without the need of cloud queries.

haslinghuis, 2024-10-02:
@mheidinger see https://github.com/betaflight/betaflight-configurator/pull/4150

Ports tab still needs to be done.

closed 2025-03-10 by haslinghuis