betaflight/betaflight-configurator #3443: W11 bf 10.9 ports stuck "Loading..."

oXeroXo, 2023-05-07:
### Describe the bug

Tried updating to 10.9 from (fully functioning 10.7).  Now I cannot get the ports to register anything.  

I believe this to be an issue with w11 compatability.  I tried this on my spare laptop with same w11 service pack updates and same issue.  

I've tried impulserc dfu driver fix with no avail. 

Reinstall 10.7 and everything works as it should.

### To Reproduce

Update to 10.9 on up to date w11 machine. 

Revert back to 10.7 and ports come back and I can connect to fc. 

(Same with several of my quads)

### Expected behavior

Ports drop down stuck on "loading"

### Configurator version

10.9

### Add any other context about the problem that you think might be relevant here

This is the only thing I have found reporting the same issue in Linux.  Could this be related to the compiled install file for windows?

https://github.com/betaflight/betaflight-configurator/issues/3336

haslinghuis, 2023-05-07:
http://betaflight.com/docs/wiki/configurator/firmware-flasher-tab#troubeshooting

oXeroXo, 2023-05-07:
Use a good quality data cable not a power cable.
Tried several same issue

USB hubs or OTG cables are sometimes needed with recent computers and in other cases they can cause issues.

Tried.  Same issue.

Try disconnecting all cables from your PC first, try rebooting, other ports, upgrade system drivers. Remove other USB connections.

Tried this.  No change.  Matter of fact 10.7 works fine with anything I haveTried this.  No change.  Matter of fact 10.7 works fine with anything I have plugged in for other devices.

Try DFU mode (use boot button on FC while plugging in, use Activate Boot Loader / DFU button in setup tab or use bl command in CLI.

Can get it into dfu mode using irc fix but won't connect and won't flash fc.

Sometimes peripherals on the flight controller such as receivers or GPS devices can hijack port communication requiring de-soldering.

Think this is a non issue due to the fact everything works perfectly fine on 10.7 

This happens on every fc I've tried plugging into anyways. I've tried 4 different setups

Linux needs configuration to allow flashing.

N/a

MacOS or Windows do not need any drivers.
If it still doesn't work try IRC Driver Fix or Zadig on Windows platform.

Irc driver fix is the only way I can have it even register anything on comm ports.  Device manager also seems to show correct driver.

oXeroXo, 2023-05-07:
[image]
[image]

haslinghuis, 2023-05-07:
Ah I know your problem. This has been solved in https://github.com/betaflight/betaflight-configurator/pull/3337

Please try latest configurator: https://github.com/betaflight/betaflight-configurator-nightlies/releases

oXeroXo, 2023-05-07:
> Ah I know your problem. This has been solved in #3337
> 
> Please try latest configurator: https://github.com/betaflight/betaflight-configurator-nightlies/releases

Dude!  This was it!!! I even tried RC5 on 10.9 thinking maybe it was fixed on one of those versions but i never tried this nightly.  Issue now resolved!  

Thanks your amazing :)

oXeroXo, 2023-05-08:
Hey so I went to start setting up the HD osd features and I noticed this 10.10 debug nightly doesn't have an osd tab.  

Ughhh this is the only reason I even decided to update this quad.  

Any idea on when the current 10.9 configurator will have this issue resolved?  I can't be the only one with this problem right now :(

closed 2023-05-08 by oXeroXo