betaflight/betaflight-configurator #3447: Betaflight 4.4.1 - unexpected full throttle

ludicek, 2023-05-08:
### Describe the bug

After changing Frsky Rx to ELRS I was flying my GepRc Crocodile baby 4 without OSD because version 4.3.1 retroactively refused any addition of DJI on the UART peripherals, so I flashed FC GepRc 722 AIO to the latest Betaflight 4.4.1, and that's when things started to happen.
Basically, after trying to start, the copter shakes horribly and within 3seconds it needs to be armed because the copter goes full throttle, shoots off somewhere sideways at full throttle. No throttle response!!! I put all the settings as I had with the previous FW.
I tried reflash again, same behavior, tried downgrade to BF 4.3.1 - there is no longer any option to add DJI to peripherals. No DJI OSD possibility event with using previous betaflight configuratos 10.8.0.etc.
This is when prop off:
https://photos.app.goo.gl/jNn8fZF3nucVgn869
In real flight it shakes horribly and after 2seconds unexpected full throttle, only one solution is immediately disarm.
No problems when flying my other models with ELRS.

### To Reproduce

impossible to use safely

### Expected behavior

no compatibility

### Configurator version

BF configurator 10.9.0

### Add any other context about the problem that you think might be relevant here

Happymodel ELRS 2,4gHz EP1 receiver, Caddx polar camera.

haslinghuis, 2023-05-09:
I guess the BNF comes with a specific tune. Did you save a diff + dump before flashing?

https://www.rcgroups.com/forums/showthread.php?3876899-GepRC-Croc-Baby-4-problems

Maybe this helps:
Painless360 dump and diff for the baby croc
https://www.dropbox.com/s/vb183syq39v9sn5/GEPRCBabyCroc4inch.zip?dl=0

haslinghuis, 2023-05-09:
Add this in CLI:

```
# serial
serial 0 1 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 30 2 115200 57600 0 115200

# aux
aux 0 0 3 1700 2100 0 0
aux 1 1 0 900 1300 0 0
aux 2 2 0 1300 1675 0 0
aux 3 46 2 1700 2100 0 0
aux 4 13 2 1700 2100 0 0

# rxfail
rxfail 7 s 1800

# master
set gyro_lowpass2_hz = 325
set dyn_lpf_gyro_min_hz = 260
set dyn_lpf_gyro_max_hz = 650
set acc_calibration = 8,1,40,1
set mag_bustype = I2C
set mag_i2c_device = 1
set mag_hardware = NONE
set baro_bustype = I2C
set baro_i2c_device = 1
set baro_hardware = NONE
set serialrx_provider = SBUS
set dshot_burst = AUTO
set dshot_bitbang = OFF
set motor_pwm_protocol = DSHOT300
set failsafe_procedure = GPS-RESCUE
set align_board_yaw = 45
set current_meter = ADC
set battery_meter = ADC
set beeper_inversion = ON
set beeper_od = OFF
set small_angle = 180
set gps_provider = UBLOX
set gps_auto_baud = ON
set gps_rescue_min_sats = 6
set osd_units = IMPERIAL
set osd_vbat_pos = 2465
set osd_flymode_pos = 2105
set osd_throttle_pos = 2137
set osd_current_pos = 385
set osd_mah_drawn_pos = 428
set osd_craft_name_pos = 9
set osd_gps_speed_pos = 2241
set osd_gps_lon_pos = 33
set osd_gps_lat_pos = 49
set osd_gps_sats_pos = 2081
set osd_home_dir_pos = 2092
set osd_home_dist_pos = 2094
set osd_altitude_pos = 2263
set system_hse_mhz = 8
set max7456_spi_bus = 2
set dashboard_i2c_bus = 1
set gyro_1_bustype = SPI
set gyro_1_spibus = 1
set name = Crocodile baby

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 91
set dyn_lpf_dterm_max_hz = 221
set dterm_lowpass2_hz = 195
set anti_gravity_gain = 5000
set feedforward_transition = 30
set iterm_relax_type = GYRO
set iterm_relax_cutoff = 20
set p_pitch = 56
set i_pitch = 65
set d_pitch = 70
set f_pitch = 75
set p_roll = 52
set i_roll = 60
set d_roll = 62
set f_roll = 75
set p_yaw = 90
set i_yaw = 70
set f_yaw = 80
set d_min_roll = 39
set d_min_pitch = 22
set d_min_boost_gain = 27

rateprofile 0

# rateprofile 0
set thr_expo = 50
set roll_rc_rate = 130
set pitch_rc_rate = 130
set yaw_rc_rate = 130
set roll_expo = 20
set pitch_expo = 22
set yaw_expo = 20
set tpa_rate = 75
set tpa_breakpoint = 1250

save
```

ludicek, 2023-05-11:
Hello,
Factory settings did not work, dron is shaking hard.
I uploaded  my saved setting which worked fine before but even this did not
help.
This is my setting with I was flying.
defaults nosave

# version
# Betaflight / STM32F7X2 (S7X2) 4.3.1 Jul 13 2022 / 03:36:10 (8d4f005) MSP
API: 1.44
# config: manufacturer_id: GEPR, board_name: GEPRC_F722_AIO, version:
33745ecc, date: 2022-11-08T04:04:44Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name GEPRC_F722_AIO
manufacturer_id GEPR
mcu_id 003800363331510c33323639
signature

# name: Crocodile4

# feature
feature -RX_SERIAL
feature GPS

# serial
serial 0 65536 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 2 2 115200 57600 0 115200
serial 3 0 115200 57600 0 115200
serial 4 0 115200 57600 0 115200

# beacon
beacon RX_LOST
beacon RX_SET

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 2 900 1300 0 0
aux 2 2 2 1300 1675 0 0
aux 3 13 1 1725 2100 0 0
aux 4 33 3 1850 2100 0 0
aux 5 34 5 1850 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 5
vtxtable powervalues 25 100 200 400 600
vtxtable powerlabels 25 100 200 400 600

# rxfail
rxfail 7 s 1800

# master
set acc_calibration = 13,10,52,1
set mag_hardware = NONE
set baro_hardware = NONE
set serialrx_provider = CRSF
set blackbox_sample_rate = 1/16
set align_board_yaw = 0
set vbat_max_cell_voltage = 436
set vbat_warning_cell_voltage = 330
set small_angle = 180
set gps_provider = UBLOX
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY
set gps_rescue_min_sats = 5
set gps_rescue_min_dth = 200
set osd_warn_batt_not_full = OFF
set osd_warn_core_temp = OFF
set osd_warn_rssi = ON
set osd_warn_link_quality = ON
set osd_warn_rssi_dbm = ON
set osd_rssi_alarm = 156
###ERROR IN diff: CORRUPTED CONFIG: osd_rssi_alarm = 156###
set osd_tim2 = 1
set osd_vbat_pos = 322
set osd_rssi_pos = 27
set osd_link_quality_pos = 2200
set osd_rssi_dbm_pos = 2169
set osd_tim_1_pos = 2282
set osd_tim_2_pos = 21
set osd_flymode_pos = 2138
set osd_ah_pos = 2126
set osd_craft_name_pos = 2506
set osd_gps_speed_pos = 2059
set osd_gps_lon_pos = 2528
set osd_gps_lat_pos = 2545
set osd_gps_sats_pos = 2106
set osd_home_dir_pos = 2159
set osd_home_dist_pos = 2063
set osd_altitude_pos = 2094
set osd_warnings_pos = 14761
set osd_avg_cell_voltage_pos = 2284
set vtx_band = 5
set vtx_channel = 7
set vtx_power = 5
set vtx_freq = 5880
set name = Crocodile4

profile 0

# profile 0
set p_pitch = 61
set i_pitch = 109
set p_roll = 58
set i_roll = 104
set p_yaw = 58
set i_yaw = 104
set simplified_pi_gain = 130

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration

So I think this might by a problem of FC not problem of Betaflight.
Thanks Ľudo, Slovakia.

sugaarK, 2023-05-24:
It looks like it’s not got rpm flitering running any reason for that? Bluejay can be used on any thing with blheliS and will give you rpm filtering.  Would make a big difference I thinks

ludicek, 2023-05-24:
Thanks, I've sent a quad to a quad service and it is already on the way
back home, so I'll let you know what was the problem.

closed 2023-06-15 by haslinghuis