betaflight/betaflight-configurator #3457: cloud build feature selection component is buggy

GTorreil, 2023-05-19:
### Describe the bug

I have consistently reproduced buggy behavior from the "tag list" component that handles feature selection for cloud builds. 

A video of the problem is availabile here : https://youtu.be/Zmzu7pTbqzc

### To Reproduce

As shown in the video, remove some features and add one (in my case , servo). 
the feature does not appear but it is selected, some text remains and there's void space in the input. 

### Expected behavior

An example of expected behavior can be found here :

https://vuetifyjs.com/en/components/autocompletes/#usage

Select : 
- [x] clearable
- [x] chips
- [x] multiple

### Configurator version

10.9.0

### Flight controller configuration

```txt
N/A
```

haslinghuis, 2023-05-19:
Often when I need to add something into Custom Defines mouse was left above - and I'm starting to type into Other Options.
Removing the ability to type in Other Options is probably the solution to this.

VitroidFPV, 2023-05-19:
I'm not sure what to fix, maybe besides the extra margin on the left. Would it be preferential to try to make the text box behave exactly as the example, or make it purely a select box?

haslinghuis, 2023-05-19:
Agree - would opt for pure select box.

GTorreil, 2023-05-20:
Hello, sorry for the late reply. 

UX bugs are very hard to describe, so I hoped my video would help. But I can say for sure that following my STRs lead to behavior that is unexpected and misleading (selected features do not appear, weird text behavior)

With that said, it is not a major roadblock so the best solution might be the most simple, here are some ideas I can think of :
- Exchange the "tag-style inputs" for a set of checkboxes
- Use a list of checkboxes that can be filtered with a filter string 

Cheers,

PR #3500 (2023-08-24): Disable search in multiple select field for Other Options field when not selected
Fixes #3457

We should not be able to type here - this part is still there - but disabling also disable the search. Disabling on field exit seems to fix the issue jumping back from typing in Custom Defines to Other Options field.

[image]
closed 2023-08-24 by haslinghuis