betaflight/betaflight-configurator #3458: Enable GPS rescue settings when GPS rescue is not selected as stage2 failsafe

catphish, 2023-05-21:
### Is your feature request related to a problem? Please describe

I wanted to configure GPS rescue on a switch, but not enable it as my stage2 failsafe. After looking at the code and testing it, I found that this is a valid and supported configuration. However, the configurator will not allow GPS rescue settings to be changed unless GPS rescue is selected as the stage2 failsafe.

It is inconvenient and confusing to have to enable GPS stage2 failsafe in order to change these settings.

In my opinion, GPS rescue is a separate feature from failsafe, and should not be conflated in the way it is at present.

### Describe the solution you'd like

I would suggest that the selection of a stage2 failsafe be decoupled from the GPS rescue settings, thus allowing GPS rescue to be configured even if it is not selected as the stage2 failsafe behaviour.

### Describe alternatives you've considered

These options could be moved to a new location, and only enabled if either stage2 failsafe GPS rescue, or mode based GPS rescue is enabled. However, I see no reason to disable them in any situation.

haslinghuis, 2023-05-24:
```
# get failsafe_procedure
failsafe_procedure = DROP
Allowed values: AUTO-LAND, DROP, GPS-RESCUE
```

catphish, 2023-05-24:
> You can change the settings in the cli any time you like

You can indeed, but it's nicer to use a GUI :slightly_smiling_face:

catphish, 2023-05-25:
> > It is inconvenient and confusing to have to enable GPS stage2 failsafe in order to change these settings.
> 
> > In my opinion, GPS rescue is a separate feature from failsafe, and should not be conflated in the way it is at present.
> 
> ```
> # get failsafe_procedure
> failsafe_procedure = DROP
> Allowed values: AUTO-LAND, DROP, GPS-RESCUE
> ```

This demonstrates that failsafe can indeed activate GPS rescue, But it's not the **only** way to activate GPS rescue.

The point of this issue is that GPS rescue can still be used when `failsafe_procedure = DROP`, but in that case it can't be configured by the configurator.

haslinghuis, 2023-05-25:
:stuck_out_tongue: Hence the added

[image]

PR #3611 (2023-11-02): Enable GPS Rescue UI if used as a mode
Fixes #3458. Currently the UI for changing GPS Rescue settings is only enabled if the failsafe stage 2 is configured as GPS Rescue. Because the GPSR settings can be changed at anytime using the CLI, it makes sense to enable the UI also if GPSR is enabled as a mode to make things easier for users.
closed 2023-11-02 by haslinghuis