betaflight/betaflight-configurator #3466: Ports tab loading indefinitely

BrandtBane27, 2023-06-03:
### Describe the bug

When this version of Betaflight is downloaded on my computer, the tab where all your computers ports are supposed to show up for you to manually select, the only selection that shows up is "Loading...", which never goes away. I have tried restarting the computer and program with all three given drivers in every combination possible and have not been able to make any progress on the issue. The only way I can resolve the bug is to download the older 10.8 version of betaflight. Which fixes this issue. However, I am still unable to connect to my flight controllers consistently on that version. Every time I wish to connect to to my drone, I have to go through hours and hours of trouble shooting just to get one to connect.  Even when I mannage to get it to work, if I go away for a day, I'm sure to have to do it all over again because it stops working again.

### To Reproduce

This happens every time I open the application, with or without drivers installed. I've also uninstalled and re-installed betalight multipule times with no success.

### Expected behavior

When this version of Betaflight is downloaded on my computer, the tab where all your computers ports are supposed to show up for you to manually select, the only selection that shows up is "Loading...", which never goes away.

### Configurator version

Betaflight 10.9

haslinghuis, 2023-06-03:
There is a problem with mdns discovery.

Fixed in [nightly configurator](https://github.com/betaflight/betaflight-configurator-nightlies/releases)

haslinghuis, 2023-06-03:
Duplicate of #3336

closed 2023-06-03 by haslinghuis