betaflight/betaflight-configurator #3478: GPS M10 Blocks usb on BF Configurator 10.10.0

Pancronos, 2023-06-09:
### Describe the bug

BFC version 10.10 (nightly) workbench testing 4.5 zulu on a cinelog 30 with a flywoo M10 micro gps, after a warm up (30 seconds on ) it gets satellite and 3d fix. Pretty good. 
[image]
The cli shows no gps ? Speed not set ?
[image]
the port is set as reccomended at 115200
[image]
After few minutes (nearly 10) BF stops responding, disconnect doesn't work, Please note that on other builds with M8 gps it all works fine.
Changing tab just crash BF
[image]
Disconnect / reconnect ends to same behaviour, only solution is to physically remove usb from pc port.
[image]

### To Reproduce

Set an fc with above setup. and follow same steps.

### Expected behavior

BF configurator  after changing gps setting should not  go crazy and block USB.

### Configurator version

2023-06-09 @22:13:01 -- OS: Windows  2023-06-09 @22:13:01 -- Configurator: 10.10.0-debug-84a0d6b

### Flight controller configuration

```txt
Entering CLI Mode, type 'exit' to return, or 'help'

# 
# Building AutoComplete Cache ... Done!
# 
# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.5.0 Jun  9 2023 / 15:16:04 (7d7b6596b) MSP API: 1.46
# config rev: 547d467

# start the command batch
batch start

board_name GEPRC_F722_AIO
manufacturer_id GEPR

# name: Cinebot30

# resources
resource MOTOR 1 C07
resource MOTOR 2 C06
resource MOTOR 3 C09
resource MOTOR 4 C08
resource PINIO 2 A01

# feature
feature -AIRMODE
feature GPS
feature TELEMETRY
feature OSD

# serial
serial 4 2 115200 115200 0 115200

# beacon
beacon RX_LOST
beacon RX_SET

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 900 1700 0 0
aux 2 2 1 1700 2100 0 0
aux 3 13 2 1700 2100 0 0
aux 4 28 1 1700 2100 0 0
aux 5 35 3 1700 2100 0 0

# master
set gyro_lpf1_static_hz = 0
set dyn_notch_count = 1
set dyn_notch_q = 600
set dyn_notch_min_hz = 150
set gyro_lpf1_dyn_min_hz = 0
set acc_lpf_hz = 10
set acc_trim_pitch = -1
set acc_trim_roll = -1
set acc_calibration = -69,-45,-20,1
set rc_smoothing_auto_factor = 250
set rc_smoothing_auto_factor_throttle = 250
set rc_smoothing_setpoint_cutoff = 6
set rc_smoothing_feedforward_cutoff = 6
set rc_smoothing_throttle_cutoff = 20
set dshot_bidir = ON
set motor_output_reordering = 2,0,3,1,4,5,6,7
set align_board_roll = 180
set align_board_yaw = 0
set vbat_max_cell_voltage = 435
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set small_angle = 180
set gps_ublox_use_galileo = ON
set gps_rescue_min_start_dist = 30
set gps_rescue_ascend_rate = 500
set gps_rescue_ground_speed = 2000
set gps_rescue_descent_dist = 200
set gps_rescue_landing_alt = 5
set gps_rescue_throttle_max = 1600
set gps_rescue_throttle_hover = 1280
set gps_rescue_sanity_checks = RESCUE_SANITY_ON
set gps_rescue_min_sats = 5
set gps_rescue_allow_arming_without_fix = ON
set gps_rescue_throttle_p = 150
set gps_rescue_throttle_i = 20
set gps_rescue_throttle_d = 50
set gps_rescue_velocity_p = 80
set gps_rescue_velocity_i = 20
set gps_rescue_velocity_d = 15
set gps_rescue_yaw_p = 40
set osd_vbat_pos = 2658
set osd_rssi_pos = 3126
set osd_rsnr_pos = 3158
set osd_tim_1_pos = 3663
set osd_tim_2_pos = 3631
set osd_flymode_pos = 2110
set osd_vtx_channel_pos = 404
set osd_crosshairs_pos = 2462
set osd_ah_pos = 2335
set osd_current_pos = 259
set osd_craft_name_pos = 2076
set osd_gps_sats_pos = 2081
set osd_warnings_pos = 2681
set osd_avg_cell_voltage_pos = 2626
set osd_battery_usage_pos = 2744
set osd_displayport_device = MSP
set osd_canvas_width = 60
set osd_canvas_height = 22
set debug_mode = GYRO_SCALED
set vcd_video_system = HD
set pinio_config = 129,1,1,1
set pinio_box = 39,40,41,42
set gyro_2_bustype = SPI
set gyro_2_sensor_align = CW0
set craft_name = Cinebot30
set pilot_name = Mat

profile 1

# profile 1
set vbat_sag_compensation = 100
set horizon_level_strength = 50
set thrust_linear = 20
set feedforward_averaging = 2_POINT
set feedforward_smooth_factor = 75
set feedforward_jitter_factor = 13

rateprofile 0

# rateprofile 0
set roll_rc_rate = 2
set pitch_rc_rate = 2
set yaw_rc_rate = 2
set roll_srate = 50
set pitch_srate = 50
set yaw_srate = 50

# end the command batch
batch end
```

### Add any other context about the problem that you think might be relevant here

Other build with M8 gps works fine and smooth. Will try to set M10 on a working build and report.

haslinghuis, 2023-06-09:
Can you try https://github.com/betaflight/betaflight/pull/12799

Pancronos, 2023-06-12:
Further tests:

BF doesn't crash even after long time i can still change tabs.
I tried 57600 no luck, Auto (reboots as 57600) both do not work # gps_info ># MON-VER acquired: false
After disconnecting usb for 2 minutes (my gps has no battery/memory) almost immediatly got 3dfix, wonderful!!

haslinghuis, 2023-06-12:
Can you press F12 for console.log for more info?

Sensor state seems be updating fine here.
Acknowledge issues with current version when GPS is attached - eventually serial / msp communication becomes unresponsive. (switching to ACM1 on Linux after restarting configurator).

haslinghuis, 2023-06-12:
Works on master with linux and windows here. Logs does not show any excessive timeouts.

Putting GPS on 115200 manually in ports tab works good for me. (AUTO always revert to 57600).
For me after connection it takes around 4 seconds before the RED or YELLOW gps icon lights up.

Using this setup:

[image]

haslinghuis, 2023-06-12:
O wow after disabling AutoConfig and reconnecting USB it totally didn't work. Reverting AutoConfig it took some extra time to come back.

Now testing with https://github.com/betaflight/betaflight/pull/12799 we definite need auto-config enabled. Without auto-config the icon is toggling on and off continues and after cold boot the GNSS device is not being recognized (resulting in icon to stay grey).

Pancronos, 2023-06-13:
I might have found the issue. 
With another FC Mamba722_2022B gps on  uart 5 it's all ok !!!
As soon as you connect:

[image]

after a 3d fix:
      
[image]

Both have same Flywoo tiny GPS module, the only difference is that with MAMBA722_2022B fc gps lights up (uart 5)  only after i plugin the battery .
with GEPRCF722 AIO (uart 5 on the manual they indicate uart3) If i connect usb and lights up, but even when i connect battery ...no gps icon. 
I don't think uart should matter though, but sounds like from where the voltage come from.

Pancronos, 2023-06-13:
i am a ignorant on this, but geprc has integrated BEC while on Mamba could not find, could that be the issue ?

haslinghuis, 2023-06-13:
UART should not matter

The M10 (bainianxing BU-1025TF) is bench tested on MAMBAG4 without other connections or LIPO power.
The M8 (beitian bn-180) sit's on a working quad with TUNERCF405 and needs LIPO to get detected.

Pancronos, 2023-06-13:
Oh well, really do not know what to say other then try another fc. I hope you get my point......one fc doesn't start gps just with USB, it needs battery, then it all works(GPS Icon). The GEPRC FC with usb starts gps but then even with battery doesn't show the GPS icon. both works!!

Pancronos, 2023-06-15:
I have tried some "core only" and several version (4.4.1 and 4.4.2), on my GEPRCF722 AIO but GPS icon doesn't show up. 
With PR#12799 it all works nice, both FC (GEPRC and Diatone) keep GPS funcionality, on the mamba F7 Aio even gps icon works. The very good part of this PR for me is that the FC's doesn't get stuck and go crazy which was the initial issue of this problem report. Will try further test on other FC and report.

haslinghuis, 2023-06-15:
Core only does not include GPS feature, only drivers.
Good to know it works with `#12799` now

Pancronos, 2023-06-24:
Foxeer M10 (M10-180) on a Speedybee F7V3 is ok, fully working, icon red and when connected
https://build.betaflight.com/api/builds/81a8c78d8c4756595843fe0095e9d2e1/log
Entering CLI Mode, type 'exit' to return, or 'help'

# 
# Building AutoComplete Cache ... Done!
# 
# status
MCU F722 Clock=216MHz, Vref=3.29V, Core temp=60degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 3419, max available: 16384
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 2 locked dma
GYRO=BMI270, ACC=BMI270, BARO=BMP280
OSD: MSP (30 x 13)
BUILD KEY: 81a8c78d8c4756595843fe0095e9d2e1 (4.5.0-zulu)
System Uptime: 15 seconds, Current Time: 2023-06-24T18:41:24.765+00:00
CPU:7%, cycle time: 316, GYRO rate: 3164, RX rate: 15, System rate: 9
Voltage: 0 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: Manufacturer 0x1d, 492544kB, 10/2016, v1.0, 'SD   '
Filesystem: Ready
GPS: connected, UART6 115200 (set to 115200), configured
     Ublox HW Version: M10, SW Version: 5.10, Proto Version: 34.10
Arming disable flags: RXLOSS CLI MSP

[image]

The only issue i have, when i enable magnetometer, heading shows dummy number and gps goes crazy, the map rotate around the center of the location point

Pancronos, 2023-07-04:
newbeedrone M10-5883 mini all ok recognized and icon visible
# status
MCU F40X Clock=168MHz (PLLP-HSI), Vref=3.26V, Core temp=57degC
Stack size: 2048, Stack address: 0x1000fff0
Configuration: CONFIGURED, size: 3748, max available: 16384
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 1 locked dma
GYRO=ICM42688P, ACC=ICM42688P, BARO=DPS310
OSD: MAX7456 (30 x 13)
BUILD KEY: 959b88c82aab8cfd2006422f5fe49c67 (4.5.0-zulu)
System Uptime: 38 seconds, Current Time: 2023-07-04T17:13:32.646+00:00
CPU:41%, cycle time: 128, GYRO rate: 7812, RX rate: 15, System rate: 9
Voltage: 1680 * 0.01V (4S battery - OK)
I2C Errors: 0
FLASH: JEDEC ID=0x00ef4018 16M
GPS: connected, UART2 115200 (set to 115200), configured
     Ublox HW Version: M10, SW Version: 5.10, Proto Version: 34.10
Arming disable flags: RXLOSS CLI MSP

[image]

Pancronos, 2023-07-04:
HGLRC M100-mini all ok recognized and icon visible [Diatone Mamba F405 mini stack]

haslinghuis, 2023-07-04:
Please test https://github.com/betaflight/betaflight/pull/12799 by adding #12799 in the commit field when flashing.

Pancronos, 2023-07-04:
All above are with PR# 12799

Pancronos, 2023-07-04:
very cheap: https://www.aliexpress.com/i/1005005367618186.html   Also works fine, they were set for NMEA so runned u-center2 with M10  gps fix (thanks RabbitAmbulance). 

As conclusion, the only one i had problem is flywoo, works but just the icon doesn't show up. Will try also that on this FC just to make sure the problem was not the fc

Pancronos, 2023-07-05:
I guess i have to apologize for opening this issue. Also the flywoo goku mini m10 works on his FC GEPRC F7. 
The issue of no icon is caused by not plugging the battery, as when i bench tested GPS was working fine just with USB 5 volts.
Adding the battery to FC the icon appeared. 
If BF Configurator could be modified to show icon even with usb 5 volts (data is in gpspassthrough) would be better and less user opening issues :)
Anyway it has been a good occasion to test some gps !!!

haslinghuis, 2023-07-05:
> If BF Configurator could be modified to show icon even with usb 5 volts (data is in gpspassthrough) would be better and less user opening issues :)

Status is obtained from firmware using MSP and is updated 4 times a second. Have reworked the live update in configurator 10.10 as the update was not working after switching [certain] tabs in configurator 10.9.

https://github.com/betaflight/betaflight-configurator/blob/9b9e5bf99ec45e0b9ec234d7b170c2427905b9fd/src/js/serial_backend.js#L717-L720

haslinghuis, 2023-07-05:
> Anyway it has been a good occasion to test some gps !!!

Please provide a list with tested GPS as we like to build a list with supported and tested GPS.

Pancronos, 2023-07-05:
FOXEER M10Q-180                                           
https://www.foxeer.com/foxeer-m10q-180-gps-5883-compass-g-507

MATEKSYS M10Q-5883                                     
http://www.mateksys.com/?portfolio=m10q-5883

NEEWBEEDRONE M10 Micro with compass       
https://newbeedrone.com/collections/all/products/newbeedrone-m10q-micro-gps-module-with-compass

FLYWOO GOKU GM10 Nano V3
https://flywoo.net/products/goku-gm10-nano-v3-gps-w-compass

HGLRC GPS-M100 MINI
https://www.hglrc.com/products/hglrc-m100_mini-gps

A-SKY AK413 ubx M10050-KB
https://www.aliexpress.com/i/3256805181303434.html?gatewayAdapt=4itemAdapt

haslinghuis, 2023-07-05:
Keeping this open for now so we can collect more data and catch non working devices

Pancronos, 2023-07-05:
A you mentioned the serial_backend change i downloaded latest configurator an tried, 
GEPRC F7 ( no battery)
[image]
gets satellites, no icons

and DIAT/MAMBAF722
[image]
No satellites, no icons

Both work with battery plugged in.