betaflight/betaflight-configurator #3479: GPS status icon does not illuminate orange when GPS is connected

catphish, 2023-06-10:
### Describe the bug

When a GPS device is connected and running, the GPS peripheral status indicator for GPS in the top-right of the configurator remains grey rather than turning orange.
This can be seen in the screenshot in #3478 and on my own build:
[image]

### To Reproduce

Connect a current build of the configurator to a current build of betaflight. Connect and configure a GPS receiver.

### Expected behavior

The GPS status indicator should turn orange to indicate communication with a GPS device.

### Configurator version

10.10.0-debug-43f1558

### Add any other context about the problem that you think might be relevant here

I am running Betaflight 4.5.0 from git master: 7d7b6596b0839c1047704d29ae2a69aec015bd3e

haslinghuis, 2023-06-10:
Still works:

[image]

[image]

Pancronos, 2023-06-12:
Further tests:
1. BF doesn't crash even after long time i can still change tabs.
2. I tried 57600 no luck, Auto (reboots as 57600) both do not work # gps_info ># MON-VER acquired: false
3. After disconnecting usb for 2 minutes (my gps has no battery/memory)  almost immediatly got 3dfix, wonderful!!

haslinghuis, 2023-06-12:
Duplicate of #3478

closed 2023-06-12 by haslinghuis