betaflight/betaflight-configurator #3488: No port appear in the list with portable version

drepou, 2023-06-19:
### Describe the bug

using the 10.9.0 standalone version i dont have any "usb" or serial port in the list so i can't connect to my drone... 
it's work with the 10.0.7 'google apps' version

### To Reproduce

use win 10 
Édition	Windows 10 Famille
Version	22H2
Installé le	‎09/‎02/‎2023
Build du système d’exploitation	19045.3086
Expérience	Windows Feature Experience Pack 1000.19041.1000.0

copie stanalone version on computer https://github.com/betaflight/betaflight-configurator/releases/download/10.9.0/betaflight-configurator_10.9.0_win64-portable.zip

run it

### Expected behavior

should list the serial port

### Configurator version

10.9.0

haslinghuis, 2023-06-19:
Enable airplane mode and restart configurator.

When port appears you need the nightly configurator otherwise need tools like IRC Driver Fix.

drepou, 2023-06-19:
> Enable airplane mode and restart configurator.
> 
> When port appears you need the nightly configurator otherwise need tools like IRC Driver Fix.

How i find a solution there is an option in the "option menu" left side called "display all serial for manufacturer and debug" activate make the prolific displayed

drepou, 2023-06-19:
use the option "display all serial for devellopement"

drepou, 2023-06-19:
How i find a solution there is an option in the "option menu" left side called "display all serial for manufacturer and debug" activate make the prolific displayed

closed 2023-06-19 by drepou