betaflight/betaflight-configurator #3490: GPS Stopped working when upgraded to 4.4.2

mactac, 2023-06-19:
### Describe the bug

I had 4.3.1 BF on my quad, and the GPS worked properly.

I upgraded to 4.4.2 and now BF will not detect the GPS.   It's set up correctly on the ports (I didn't change it), but on the config tab when I turn on the GPS and tell it what type of GPS I have, then save - it goes back to off.

I'm using a BN-880 GPS
Tmotor F7 Board

### To Reproduce

Upgraded to 4.4.2

### Expected behavior

GPS should work as it did in the previous version

### Configurator version

10.9.0

### Flight controller configuration

```txt
# version
# Betaflight / STM32F7X2 (S7X2) 4.4.2 Jun  4 2023 / 22:23:47 (23d066d08) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name TMOTORF7
manufacturer_id TMTR

# feature
feature -TELEMETRY
feature -AIRMODE

# serial
serial 1 64 115200 57600 0 115200
serial 3 131073 115200 57600 0 115200
serial 4 2 115200 57600 0 115200

# beeper
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -DISARM_REPEAT
beeper -ARMED
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# aux
aux 0 0 1 1700 2100 0 0
aux 1 1 0 900 1300 0 0
aux 2 2 0 1300 1700 0 0
aux 3 28 0 1700 2100 0 0
aux 4 35 2 1700 2100 0 0
aux 5 39 4 900 1300 0 0

# master
set gyro_lpf1_static_hz = 150
set gyro_lpf2_static_hz = 300
set dyn_notch_count = 2
set gyro_lpf1_dyn_min_hz = 150
set gyro_lpf1_dyn_max_hz = 300
set acc_calibration = 15,9,-9,1
set baro_hardware = NONE
set rssi_channel = 12
set rc_smoothing_auto_factor = 35
set blackbox_sample_rate = 1/2
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 3
set failsafe_throttle_low_delay = 200
set failsafe_procedure = GPS-RESCUE
set vbat_max_cell_voltage = 440
set vbat_min_cell_voltage = 320
set ibata_scale = 152
set small_angle = 180
set simplified_gyro_filter_multiplier = 60
set osd_rssi_alarm = 40
set osd_cap_alarm = 2600
set osd_alt_alarm = 500
set osd_vbat_pos = 2080
set osd_rssi_pos = 3152
set osd_tim_2_pos = 2560
set osd_current_pos = 3566
set osd_mah_drawn_pos = 3598
set osd_gps_speed_pos = 2528
set osd_gps_lon_pos = 2624
set osd_gps_lat_pos = 3656
set osd_gps_sats_pos = 3120
set osd_home_dir_pos = 2073
set osd_home_dist_pos = 2075
set osd_flight_dist_pos = 3088
set osd_altitude_pos = 18880
set osd_warnings_pos = 14645
set osd_avg_cell_voltage_pos = 2048
set osd_efficiency_pos = 2112
set osd_displayport_device = MSP
set debug_mode = GYRO_SCALED
set vcd_video_system = HD
set pinio_box = 39,0,0,0
set rpm_filter_harmonics = 2

profile 0

# profile 0
set dterm_lpf1_dyn_min_hz = 90
set dterm_lpf1_dyn_max_hz = 180
set dterm_lpf1_static_hz = 90
set dterm_lpf2_static_hz = 180
set pidsum_limit = 1000
set pidsum_limit_yaw = 1000
set p_pitch = 59
set i_pitch = 69
set d_pitch = 72
set f_pitch = 215
set p_roll = 57
set i_roll = 66
set d_roll = 58
set f_roll = 206
set p_yaw = 57
set i_yaw = 66
set f_yaw = 206
set d_min_roll = 58
set d_min_pitch = 72
set thrust_linear = 20
set dyn_idle_min_rpm = 40
set simplified_master_multiplier = 150
set simplified_i_gain = 65
set simplified_d_gain = 130
set simplified_pi_gain = 85
set simplified_dmax_gain = 0
set simplified_feedforward_gain = 115
set simplified_pitch_d_gain = 110
set simplified_dterm_filter_multiplier = 120

rateprofile 0

# rateprofile 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 150
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 40

# end the command batch
batch end
```

haslinghuis, 2023-06-19:
Please provide support ID.

See also: https://github.com/betaflight/betaflight-configurator/issues/3478

mactac, 2023-06-19:
> Please provide support ID.
> 
> See also: #3478

Support ID?  I do not understand

haslinghuis, 2023-06-19:
There should be a button in CLI tab.

mactac, 2023-06-19:
> Please provide support ID.
> 
> See also: #3478

#3478 Does not seem to be the same issue.

haslinghuis, 2023-06-19:
Without support ID there is not enough info. Did you add GPS in build settings etc?

If there is not support button in CLI checkout https://github.com/betaflight/betaflight-configurator-nightlies/releases

mactac, 2023-06-23:
Update:   So, I re-flashed BF, took the diff from above (the one I copied from this very quad), and saved it, and now it works.  So clearly something weird happened with my BF install.   It's strange because this quad has been working for a long time, and just stopped working when I flashed to the new BF.  All settings exactly the same, it just now works for some reason.

closed 2023-06-25 by haslinghuis