betaflight/betaflight-configurator #3501: Migrate OSD Preview Panel to Vue

BoldBigflank, 2023-06-27:
### Is your feature request related to a problem? Please describe

It seems the goal is to get this project moved over to Vue at some point. Some simple components have been added to the project, so making more will require no more packages. I think the OSD Tab is one of the easier ones to build and could work as a template for the rest. The first step toward that would be to make the Preview panel a Vue component.

### Describe the solution you'd like

Preview panel of the OSD tab implemented as a Vue component.
Panel reacts to updates from outside the component, like active preview and selected items.
Objects in the panel properly react to keyboard controls and mobile pointer drag/drop.
Storybook tests ensure the component acts as intended

### Describe alternatives you've considered

Could leave it alone. It works now, warts and all.
Could be more meticulous about fixing issues that it has, such as mobile drag/drop and update latency. It's an option, and wouldn't necessarily mean that we couldn't convert it to Vue or another framework later.

### Other information

I intend to do the work for this task, so I would appreciate others' feedback on methodology or tips for getting it going.

haslinghuis, 2023-06-27:
@chmelevskij

chmelevskij, 2023-06-29:
Looks great. Happy to review/help out.

I think starting off with storybook is a good shout, could create smaller PRs for review and when it comes time to connect to the whole event system happy to help.

closed 2024-06-23 by haslinghuis