betaflight/betaflight-configurator #3514: Setup and Configuration tabs will not save when using AT32 FC

robhaswell, 2023-07-15:
### Describe the bug

With the following combinations:

* BETAFPVF435 or NEUTRONRCF435
* Apple M2 16" MBP
* `master` betaflight and nightly configurators

I have an issue where changes to the Ports or Configuration tabs do not save. Upon pressing Save, a warning is generated in the console and the connection becomes unresponsive, until the FC is replugged. I have experienced this with two AT32 FCs and this issue has been present for me in nightly configurator builds since I got these a few weeks ago.

What does work:

* These tabs work with STM32 FCs.
* These tabs work on AT32 using my Windows machine. The BF developer Un!t also has tested this on Linux.
* Other tabs work as expected - Presets, Modes, OSD and CLI are the ones I have used.
* Setting the values on these tabs using the CLI also works.

Here is a screenshot of the error, in this situation I tested making two changes to Modes before triggering the bug:

[image]

### To Reproduce

1. Use the hardware configuration above.
2. Plug in the FC.
3. Navigate to the Configuration tab.
4. Enter " 2" after the pilot name.
5. Press save.
6. Observe the bug.

### Expected behavior

Settings are saved and the FC reboots.

### Configurator version

10.10.0-debug-4d42a8f

### Flight controller configuration

```txt
# diff all

# version
# Betaflight / AT32F435G (A435) 4.5.0 Jul 14 2023 / 15:02:10 (1485438c4) MSP API: 1.46
# config rev: 9c4dd3b

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name NEUTRONRCF435
manufacturer_id NERC
mcu_id 10153a22054740000d94870c
signature

# name: OpenRacer Orange

# feature
feature TELEMETRY
feature OSD

# serial
serial 0 131073 115200 57600 0 115200
serial 1 64 115200 57600 0 115200

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 1700 2100 0 0
aux 2 2 1 1300 1700 0 0
aux 3 13 2 1300 1700 0 0
aux 4 26 0 1700 2100 0 0
aux 5 35 2 1700 2100 0 0

# adjrange
adjrange 0 0 3 900 2100 12 3 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY    0    0    0    0    0    0    0    0
vtxtable band 2 BOSCAM_B B FACTORY    0    0    0    0    0    0    0    0
vtxtable band 3 BOSCAM_E E FACTORY    0    0    0    0    0    0    0    0
vtxtable band 4 FATSHARK F FACTORY    0 5760    0 5800    0    0    0    0
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM     0    0    0    0    0    0    0    0
vtxtable powerlevels 3
vtxtable powervalues 14 23 0
vtxtable powerlabels 25 200 0

# rxfail
rxfail 6 s 1500

# master
set gyro_lpf1_static_hz = 0
set dyn_notch_count = 2
set dyn_notch_q = 450
set dyn_notch_min_hz = 125
set dyn_notch_max_hz = 650
set gyro_lpf1_dyn_min_hz = 0
set gyro_lpf1_dyn_max_hz = 0
set acc_lpf_hz = 10
set acc_calibration = -1,-97,34,1
set rc_smoothing_auto_factor = 25
set rc_smoothing_auto_factor_throttle = 25
set motor_kv = 2400
set dshot_bidir = ON
set motor_output_reordering = 2,3,0,1,4,5,6,7
set align_board_yaw = -90
set vbat_min_cell_voltage = 340
set vbat_scale = 111
set yaw_motors_reversed = ON
set small_angle = 180
set simplified_gyro_filter = OFF
set report_cell_voltage = ON
set osd_vbat_pos = 385
set osd_rssi_pos = 353
set osd_link_quality_pos = 2082
set osd_rssi_dbm_pos = 377
set osd_tim_1_pos = 54
set osd_tim_2_pos = 3489
set osd_flymode_pos = 374
set osd_throttle_pos = 2614
set osd_vtx_channel_pos = 3112
set osd_crosshairs_pos = 280
set osd_craft_name_pos = 393
set osd_pilot_name_pos = 2549
set osd_warnings_pos = 14675
set osd_avg_cell_voltage_pos = 2475
set osd_disarmed_pos = 429
set osd_sys_vtx_temp_pos = 2114
set osd_stat_bitmask = 5092
set osd_displayport_device = MSP
set osd_canvas_width = 50
set osd_canvas_height = 18
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 2
set vtx_power = 1
set vtx_freq = 5695
set vcd_video_system = HD
set rpm_filter_fade_range_hz = 100
set craft_name = OpenRacer Orange
set pilot_name = Mopatop

profile 0

# profile 0
set dterm_lpf1_dyn_expo = 7
set iterm_relax_cutoff = 20
set iterm_limit = 500
set pidsum_limit_yaw = 1000
set throttle_boost = 2
set p_pitch = 41
set i_pitch = 89
set d_pitch = 40
set f_pitch = 137
set p_roll = 38
set i_roll = 81
set d_roll = 35
set f_roll = 125
set horizon_level_strength = 50
set d_min_roll = 21
set d_min_pitch = 23
set feedforward_averaging = 2_POINT
set feedforward_smooth_factor = 87
set feedforward_jitter_factor = 2
set feedforward_boost = 18
set feedforward_max_rate_limit = 100
set simplified_pids_mode = RP
set simplified_i_gain = 120
set simplified_d_gain = 70
set simplified_pi_gain = 85
set simplified_dmax_gain = 200
set simplified_feedforward_gain = 105
set simplified_pitch_pi_gain = 105
set tpa_rate = 70
set tpa_breakpoint = 1250

profile 1

profile 2

profile 3

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```

### Add any other context about the problem that you think might be relevant here

Un!t has been helping me with this, and suggested desoldering the FC, which I did - no change.

haslinghuis, 2023-07-15:
Please check build: https://github.com/betaflight/betaflight-configurator/pull/3511

haslinghuis, 2023-07-15:
Also might be relevant: https://github.com/betaflight/betaflight-configurator/pull/3512 as code 37 is `MSP_SET_FEATURE_CONFIG`

robhaswell, 2023-07-15:
Thanks. I'm racing tomorrow so don't wanna change anything now. But will test when I get back.

robhaswell, 2023-07-18:
> Please check build: #3511

This has the same issue.

robhaswell, 2023-07-18:
> Also might be relevant: #3512 as code 37 is `MSP_SET_FEATURE_CONFIG`

This also has the same issue.

haslinghuis, 2023-07-18:
As it works on Linux this issue will be hard to debug without access to an Apple M2 16" MBP.

zuidspookFPV, 2023-07-29:
I have the same problem I can’t save and reboot in ports and configuration tab in motors and receiver it works. I tested this on 2 different Mac’s and multiple nighty builds but to no luck, both the Mac’s where intel Mac’s but different models. I was able to configure it with my old android phone so now I can use the bt nano 3 module to configure it with my iPhone

sugaarK, 2023-07-29:
I have an intel Mac not an m2

robhaswell, 2023-07-29:
> I have an intel Mac not an m2

It seems that the issue is present on Intel macs as well.

luttesudo, 2023-07-31:
I had the same problem with my m1 max

Prajwal-Koirala, 2023-12-27:
**Data-driven analysis of Betaflight Configurator settings save issue on Apple M2 Macs:**

I've been digging into this issue and have some data-driven insights to share:

* **Hardware:** Apple M2 16" MBPs with nightly builds and master configurators appear to be the culprit.
* **Steps to replicate:**
    1. Use an AT32 FC.
    2. Enter "2" after the pilot name in the Configuration tab.
    3. Press save.
* **Expected behavior:** Settings saved, FC reboots.
* **Actual behavior:** Warning generated, connection unresponsive.
* **Interestingly:**
    * This issue is specific to AT32 FCs on Apple M2 Macs.
    * STM32 FCs, AT32 FCs on Windows, and other tabs like Presets, Modes, OSD, and CLI are not affected.

**Possible causes:**

* Incompatibility between Apple M2 architecture and Betaflight's saving routines for AT32 FCs.
* Bug in nightly builds or master configurators related to saving on Apple M2 Macs.

**Next steps:**

* Can we reproduce this issue on other Apple M2 Macs with different FCs?
* Are there any logs from Betaflight Configurator that might shed light on the problem?
* Have any workarounds been discovered for saving settings on AT32 FCs with Apple M2 Macs?

By sharing this data and analysis, I hope we can get closer to resolving this issue.

I'm also happy to help with further testing or investigation in any way I can.

Thanks,

Prajwal Koirala

robhaswell, 2024-01-10:
I can confirm that this issue is still present in RC2.

alteman, 2024-01-19:
FWIW this reproes on an M1 MBP.

nerdCopter, 2024-01-29:
could potentially invoke `os.release()` to to get os versions and set differing NWjs versions, but such would be a painful mess.

haslinghuis, 2024-01-29:
@robhaswell did you try https://github.com/betaflight/betaflight-configurator/pull/3695 ???

Yamboss, 2024-05-16:
im having the same problems i was sure it was just a run of bad boards- thank you for letting me know its my mac- ill use a PC for this one again huge thanks !

haslinghuis, 2024-06-23:
@robhaswell can you try https://app.betaflight.com with chromium / chrome browser?

robhaswell, 2024-06-26:
@haslinghuis I'm afraid it still doesn't work on https://app.betaflight.com/

[image]

Yamboss, 2024-06-26:
> Was kind of expecting that. Issue is an combination of firmware and
> platform used.

thanks for the update i did had to use windows unfortunately...
best of luck !

haslinghuis, 2025-05-05:
@robhaswell @Yamboss @zuidspook please test #4455

PR #4455 (2025-05-12): Fix AT32 Configuration can't be saved on MacOS
Can fix #3514, but not good.

## Summary by CodeRabbit

- **New Features**
  - Improved compatibility for macOS users when connecting to certain USB devices by automatically adjusting how data is sent.
- **Bug Fixes**
  - Enhanced data transmission reliability for specific devices on macOS platforms.
closed 2025-05-12 by haslinghuis