betaflight/betaflight-configurator #3518: configurator does not show any icons

TheIsotopes, 2023-07-19:
### Describe the bug

see on screenshot. icon in taskbar and in left upper corner don't show.

[image]

### To Reproduce

use latest configurator

### Expected behavior

display of the icon

### Configurator version

10.10

haslinghuis, 2023-07-19:
Any NW.js version above 0.72 seems to handle icon differently.

haslinghuis, 2023-07-19:
@chmelevskij @McGiverGim any idea here (besides revert the change)

closed 2023-08-01 by haslinghuis