betaflight/betaflight-configurator #3520: Visualize settings in configurator (different versions / no drone connected)

FPVSlice, 2023-07-19:
### Is your feature request related to a problem? Please describe

Quite often I want to visualize/compare my settings/profiles in Betaflight without connecting my drone.
It would also be nice to be able to do this with older dumps so i can compare this with my existing (latest) config.
This would mean i don't have to alter old config files for the configurator to read them properly if i want to test/troubleshoot differences or whatever.

Maybe this would be less useful for the veterans among us, but sometimes I just want to check what certain old values would translate to in the GUI. 
Ofcourse it would be great to be able to do this side by side visually. (so the ability to open your drone config and compare it with an older config)

### Describe the solution you'd like

Some kind of ability to load (old) betaflight dumps without connecting your drone.
This alone would be great, sometimes i just want to fiddle around in betaflight to visualize certain settings instead of just looking at a text file, that honestly is pretty difficult to read/compare sometimes.
Also make it possible to leave it open while you can tinker on your (actual) drone config.
I think the cherry on top would be some kind of diff function that would summarize the differences?

### Describe alternatives you've considered

Maybe there is a feature like this out in the wild? Maybe some browser simulation..? 
Other than that, I don't know.

haslinghuis, 2023-07-19:
Look in options tab. You might want to enable virtual mode.

closed 2023-07-19 by haslinghuis

TimboSliceOps, 2023-07-21:
> Look in options tab. You might want to enable virtual mode.

Hey, I had a look and tried testing some stuff.
It doesn't seem possible to load my configs via CLI or presets.
Can this be reopened then please?

haslinghuis, 2023-07-21:
See also #3523